MotorolaMobilityLLC/kernel-msm #8: Modification of stock boot image fails

tobiasjakobi, 2015-01-08:
Hello,

I'm trying to do the following on a Moto G (2014 / 2nd gen), which should be 'titan'. I want to modify the content of the initrd.

I've proceeded by extracting the content of the boot partition and have isolated the initrd portion from it. Unpacked it, did my modification (small change to the fstab.qcom), and repacked the whole thing. I padded the initrd such that the size exactly matches the current one and then I integrated the whole thing into the boot image again.

I flashed the image to boot via fastboot, but fastboot console just tells me 'hab check failed for boot'. I've analyzed the structure of the boot image and AFAIK there is not checksumming / signature stuff going on there, so this should work. My guess is that the bootloader still does some checksum validation (the 'hab check'). Sadly I haven't found any concrete information about what this 'hab check' constitutes.

Since Motorola folks are frequenting here, I thought I could pose this question here. So anything I'm missing in terms of boot image modification?

With best wishes,
Tobias

PS: Bootloader is unlocked.

ccfries, 2015-01-09:
What was your mkbootimg command line?

(That is a surprising error, wondering if some offset was just plain wrong)

tobiasjakobi, 2015-01-09:
I didn't use mkbootimg, just plain dd to position the initrd at the same position in the image.

At first I disassembled the image with abootimg, but I then realized that the device wouldn't even boot the reassembled version (no modifications to the parts at all, just disassembly followed by assembly).

dhacker29, 2015-01-09:
I would recommend compiling the whole boot.IMG but if not you need to use a tool that handles the dtb like https://github.com/xiaolu/mkbootimg_tools

tobiasjakobi, 2015-01-09:
Thanks David for the link, I'm going to give them a try.

ccfries, 2015-01-09:
If you grew the ramdisk slightly, maybe you blew over the device tree that
lands after the ramdisk, or something like that.  Good luck.

k2wl, 2015-01-09:
use archi's kitchen to unpack and repack the boot.img
it works perfect!!

tobiasjakobi, 2015-01-09:
@ccfries: No, like I said the target initrd has exactly the same size as the source initrd, so nothing except the initrd got overwritten in the process. My impression is that despite anything you read on the net about the boot image header, the bootloader in fact does check the SHA1 stored there.

Anyway, I gave the mkbootimg variant linked by David a try, and while these really produce a working image (it boots fine, that is), the phone becomes unuseable afterwards (SIM functionality, WiFi, etc. simply stops working).

savoca, 2015-01-10:
@tobiasjakobi You can take a look at Qualcomm's lk repository for the source on their bootloader, but most OEMs choose their own route as far as boot verification goes. If you'd like, you can compile your own lk build and replace a standard boot image's zImage with the lk binary. Make sure you point the primary boot path to a new partition for your actual boot image, and that you append the correct cmdline to the new boot image. Then you may be able to do whatever it is you intend to do.

https://www.codeaurora.org/cgit/quic/la/kernel/lk

As for the broken wlan/radio, I've only seen that when packing the ramdisk with the 'mkbootfs' binary. Try this command instead.

find . | cpio -o -H newc | gzip > "/path/to/output/initramfs.img"

tobiasjakobi, 2015-01-16:
Thanks for all the help, but I now solved my root issue with a different approach. So I have a microSD in the Moto G, which I wanted to use with ext4. But it turns out that the automounter is hardcoded to vfat (and not 'auto'). The partition is not mounted and I also get a notice each time that the microSD has an 'unsupported' filesystem (which is nonsense, since the kernel already uses ext4 for some of the mounts).

Anyway, I've now proceeded in the following way. Created the first partition on the microSD as vfat (2GB of 16GB) and allocated the rest to a second partition (ext4). The existence of the first partition removes the annoying notice. To mount the second partition I've just modified the post-boot initscripts to create a new mountpoint in /mnt and then mount it there.

I'm mostly storing music there, and this works perfectly fine in the two applications I use for playback (Rockbox and VLC).

closed 2015-01-16 by tobiasjakobi