MotorolaMobilityLLC/kernel-msm #9: Build moto G 2014 kernel with usb serial and audio driver support

pirpi, 2015-02-10:
Even I have downloaded kernel-msm-MMI-KXB21.85-24.tar.gz for moto G 2014. Then I have build that kernel after adding usb serial driver. But size is coming lesser than required. Anyhow I managed to run the newly created boot.img. But now my sound is not working. No sound at all from the device. It is showing following error when I check with dmesg.

" failed to instantiate card msm8226-tapan9302-snd-card : -22 " and " failed to init SLIMBUS_0_RX : -22 "

Could you please help me with this?

Thanks 
pirpi

somcom3x, 2015-04-13:
Out of curiosity, do you have the card enabled in the defconfig?

awheeler-mmi, 2017-09-27:
This is an old issue.  We don't actively support experimentation, but do encourage it!! I recommend checking out XDA.

closed 2017-09-27 by awheeler-mmi