MotorolaMobilityLLC/kernel-msm #12: Kernel source for falcon?

KingKaminari, 2015-02-23:
When will you release the kernel source for the Moto G 2013 (falcon) running Android Lollipop?

Firmware build versions are LXB22.46-xx

somcom3x, 2015-02-23:
My guess is that they are here: https://github.com/MotorolaMobilityLLC/kernel-msm/tree/lollipop%E2%80%945.0.2-release

KingKaminari, 2015-02-23:
Maybe? The device isn't specified. The only device-specific kernel released here is for victara (Moto X 2014).

somcom3x, 2015-02-23:
They are all unified (condor/falcon/peregrine/titan/victara) inside of the
kernel. They have a different tracking branch for the victara though.
Motorola delays the source pushes for quite a while now, maybe they'll
eventually push it with the L rollout.

tags07, 2015-02-23:
Sources for Moto G, Moto E and Moto G(2nd Gen) are all posted on branch: lollipop—5.0.2-release   and TAG : MMI-LXB22.46-28

Sources for Moto X(2nd Gen) is posted on branch:  lollipop-5.0.2-release-victara
TAG: MMI-LXE22.46-25

closed 2015-02-23 by tags07

KingKaminari, 2015-02-24:
Ah, okay then. Thanks @tags07!

cybojenix, 2015-03-02:
@tags07 I'm messing with the moto e 2nd gen atm. found the dump commands in fastboot, but it requires the moto version of fastboot (located here http://goo.gl/Qyzg2L )

```
$ sudo fastboot oem partition dump recovery
(bootloader) Latest Motorola fastboot required, download from: 
(bootloader) http://goo.gl/Qyzg2L
(bootloader) 
FAILED (remote failure)
```

would you be able to provide it please?

cybojenix, 2015-03-02:
Scratch that. I've got one that supports it, though it's limited (probably due to the "allow unlock" flag not being set [it resets after every reboot]). 

Would you be able to send the complete boot image please? Dm-verity means I can no longer boot

ccfries, 2015-03-02:
For future reference you can get windows/darwin/linux fastboot.exe from a recovery image.
https://motorola-global-portal.custhelp.com/app/standalone/bootloader/recovery-images

That being said, I don't see a falcon recovery image posted.

-Chris