MotorolaMobilityLLC/kernel-msm #16: unknown baseband issues with Victara LXE22.46-11

faux123, 2015-04-08:
Hi,

I've downloaded the latest kernel source code and have rebuild the kernel using moto's kernel sources and I'm having issues with unknown baseband?  Any tip on resolving this issue?  Thanks.

ccfries, 2015-04-08:
Did you change the ramdisk at all?

Does /firmware and /persist mount? Any selinux denials for /firmware or
/persist?

faux123, 2015-04-12:
@ccfries yes, there are tons of avc denials for /firmware /persist etc..  How do we disable selinux or add the new selinux into custom kernels?

ccfries, 2015-04-13:
Not sure whether to respond with an lmgtfy link or
http://stopdisablingselinux.com/

faux123, 2015-04-14:
@ccfries Okay, thanks for the link :p  I'm not trying to disable selinux perse, but rather how do I incorporate selinux into the motorola kernel.  The issue I'm running into that since we don't have the device tree for victara, I'm not able to generate all the selinux parts for our own kernel builds.

ccfries, 2015-04-14:
All the sepolicy configuration stuff is in the ramdisk.

faux123, 2015-04-14:
@ccfries how about the kernel module hashes and such.. do you have tools to generate them? Once I touched the ramdisk, then selinux will be complaining again

awheeler-mmi, 2017-09-27:
Old issue.

closed 2017-09-27 by awheeler-mmi