MotorolaMobilityLLC/kernel-msm #17: XT1079 source code please.

victortan1972, 2015-04-17:
Please update kernel source code for XT1079 and tag it, please.

luca020400, 2015-04-17:
https://github.com/MotorolaMobilityLLC/kernel-msm/tree/lollipop%E2%80%945.0.2-release
It uses the same branch as Moto G/Moto G LTE/MOTO G 2014

gtx1049, 2015-04-21:
@luca020400  I compiled this kernel branch "lolipop 5.0.2 release" and flash it into XT1079
But the wifi module and SIM don't work, also, I cant get baseband info
The log said "qmi" module has some problem. 
like "E/QC-QMI  (  331): qmi_client [331]: unable to connect to server"
I dont know how to fix it.
By the way, XT1079's default build version is "LXB22.99-7.1"

lucasauditoree, 2015-05-12:
como faz pre instalar : o  kernel no moto G

luk1337, 2015-05-12:
@lucasauditoree English?

lucasauditoree, 2015-05-12:
How does pre installed: the kernel in moto G

luk1337, 2015-05-12:
@lucasauditoree I don't really understand your question

lucasauditoree, 2015-05-12:
how will I do to install the kernel to revive Moto G xt1033 have to create the kernel that the link aii

efgowacki, 2015-05-12:
We are looking for the MBM-CI 5.0 file for MOTOROLA MOTO G 2013 devices (XT1032 / XT1033) HARDBRICKED. With these files will be possible to get it? May indicate how or who to?

lucasauditoree, 2015-05-13:
it also wonder

victortan1972, 2015-05-14:
MMI-LXB22.99-24 is ready now. It work on my XT1079 (Moto G 2014 Chinese version).

closed 2015-05-14 by victortan1972

lucasauditoree, 2015-05-14:
will be to work on my bike g 2013 that hardbrick