MotorolaMobilityLLC/kernel-msm #20: Non-ASCII character in branch name

stucki, 2015-06-08:
The branch https://github.com/MotorolaMobilityLLC/kernel-msm/tree/lollipop%E2%80%945.0.2-release does not contain a plain ASCII dash but some special character instead.

This breaks "repo sync" on my system (and probably on others, too). Can you rename it?

luca020400, 2015-06-09:
I don't get your problem , anyway https://github.com/MotorolaMobilityLLC/kernel-msm/tree/lollipop—5.0.2-release works

stucki, 2015-06-09:
It's not a dash and I wanted to report that. If it works for you then fine, but it failed on my system and it was hard to find the cause for it.

```
➜  msm8226 $ git:(upstream) git remote -v
MotorolaMobilityLLC	https://github.com/MotorolaMobilityLLC/kernel-msm.git (fetch)
MotorolaMobilityLLC	https://github.com/MotorolaMobilityLLC/kernel-msm.git (push)
➜  msm8226 $ git:(upstream) find .git/refs/ | egrep -v "^.git/refs/[/A-Za-z0-9_\.-]*$"
.git/refs/remotes/MotorolaMobilityLLC/lollipop—5.0.2-release
```

luk1337, 2015-06-09:
@stucki you can clone with MMI-LXB22.99-24 instead.

stucki, 2015-06-09:
Thanks for the sugggestion, but I can work around this of course (just delete the ref manually). I'm reporting it here so that it will hopefully be fixes and in the future nobody will run into the same problem again...

luk1337, 2015-06-09:
anyway MMI-LXB22.99-24 is a bit newer than lollipop—5.0.2-release.

stucki, 2015-06-09:
Right. I didn't use it anyway. However, when you run ```git fetch``` then it will create references for all branches, including the old broken one. So I don't need it, that's also why I could delete it as a workaround.
Still, the problem is that it will be added again on next fetch...

ccfries, 2015-06-09:
Yes, it's a valid bug, it'll get cleaned up.

closed 2015-09-30 by tags07