MotorolaMobilityLLC/kernel-msm #21: Moto E conor is not booting with condor 4.4.2 mr2

amitg2291632, 2015-06-11:
Hi,

I am trying to build condor kernel 4.4.2 mr2 release 
after that i am getting error ERROR: modpost: Found 2 section mismatch(es). To see full details build your kernel with: 'make CONFIG_DEBUG_SECTION_MISMATCH=y'

and when i am building with below options 
make  -C kernel_mr2/kernel-msm ARCH=arm  CROSS_COMPILE=$cross  O=$kernel_out_dir defoldconfig
make  -C kernel_mr2/kernel-msm ARCH=arm  CROSS_COMPILE=$cross  O=$kernel_out_dir headers_install
make  -C kernel_mr2/kernel-msm ARCH=arm  CROSS_COMPILE=$cross  O=$kernel_out_dir CONFIG_NO_ERROR_ON_MISMATCH=y
make  -C kernel_mr2/kernel-msm ARCH=arm  CROSS_COMPILE=$cross  O=$kernel_out_dir dtbs
make  -C kernel_mr2/kernel-msm ARCH=arm  CROSS_COMPILE=$cross  O=$kernel_out_dir modules
make  -C kernel_mr2/kernel-msm ARCH=arm  CROSS_COMPILE=$cross  O=$kernel_out_dir INSTALL_MOD_PATH=$PWD/out/target/product/generic modules_install

After that when i am flashing the kernel its not booting .

But when i trying same for kitkat-4.4.4-release-condor-repw its booting but wifi not working.

Please let me know what is wrong in mr2 or am i doing anything wrong.
Let me know for any more information.

Thanks & Regards,
Amit Garg

luca020400, 2015-06-11:
This is the wrapper , it "broke" the build if there are new warnings
For WiFi probably ou need to build prima too

awheeler-mmi, 2017-09-28:
Old issue, but, yah, you need to rebuild the wlan driver along with the kernel.

closed 2017-09-28 by awheeler-mmi