MotorolaMobilityLLC/kernel-msm #24: Release XT1053 LPA23.12-15 source Please.

socraticbliss, 2015-07-05:
Hey guys, I'm just gonna basically copy paste the same format as another previous request for another device, thanks again for your support :)

Please release the LP source for Moto X 2013 Developer Edition:

Baseband version: MSM8960PRO_BP_23255.138.89.00R
Kernel version: 3.4.42-g030aaa2 dbbuild@ca88lnxdroid14 #1 Sat May 9 02:01:57 PDT 2015
System version: 222.21.15.ghost_row.Retail.en.US tmo
Build number: LPA23.12-15

Thanks :)

socraticbliss, 2015-08-01:
Any update guys? The OTA has been available for quite a while now, along with the download directly from Google Drive

socraticbliss, 2015-09-10:
I see that it's been at least referenced, so there is some relief

Is there any statement on the progress on this? Or are you going to release version that comes with the Stagefright patch?

socraticbliss, 2015-09-18:
Revised the Kernel version to reflect the recent Stagefright patch, I hope this can be resolved soon!

Hasuris, 2015-09-29:
"You may copy and distribute the Program (or a work based on it, under Section 2) in object code or executable form under the terms of Sections 1 and 2 above provided that you also do one of the following:

    a) Accompany it with the complete corresponding machine-readable source code, which must be distributed under the terms of Sections 1 and 2 above on a medium customarily used for software interchange; or, "[...]

In case you guys over at Motorola aren't familar with GPLv2, I cited the section with the part, that says you have to ACCOMPANY any software you release under GPLv2 with the source code. It's been months now.

Pretty please?

tags07, 2015-09-29:
Thank you for bringing to our attention. I am checking this.. will keep you updated.

tags07, 2015-09-29:
We'll be pushing all the commits leading up to the latest shipping release: LPA23.12-39.7 
It should go out in the next push.

darkobas, 2015-09-29:
what about x play?  :)

tags07, 2015-09-30:
@Hasuris:  We do know GPLv2 a bit :) ....let me list the entire Section2:

```
3. You may copy and distribute the Program (or a work based on it, under Section 2) in object code or executable form under the terms of Sections 1 and 2 above provided that you also do one of the following:

a) Accompany it with the complete corresponding machine-readable source code, which must be distributed under the terms of Sections 1 and 2 above on a medium customarily used for software interchange; or,
b) Accompany it with a written offer, valid for at least three years, to give any third party, for a charge no more than your cost of physically performing source distribution, a complete machine-readable copy of the corresponding source code, to be distributed under the terms of Sections 1 and 2 above on a medium customarily used for software interchange; or,
c) Accompany it with the information you received as to the offer to distribute corresponding source code. (This alternative is allowed only for noncommercial distribution and only if you received the program in object code or executable form with such an offer, in accord with Subsection b above.)
The source code for a work means the preferred form of the work for making modifications to it. For an executable work, complete source code means all the source code for all modules it contains, plus any associated interface definition files, plus the scripts used to control compilation and installation of the executable. However, as a special exception, the source code distributed need not include anything that is normally distributed (in either source or binary form) with the major components (compiler, kernel, and so on) of the operating system on which the executable runs, unless that component itself accompanies the executable.

If distribution of executable or object code is made by offering access to copy from a designated place, then offering equivalent access to copy the source code from the same place counts as distribution of the source code, even though third parties are not compelled to copy the source along with the object code.
```

We do provide an offer statement with each product and you have the rights to request the source code if its not posted. We try to do our best to post them in time..sometimes things skip our radar or is lower in priority list to get them posted. Please be assured that it will get posted. We are close to catching up on all L posting Moto X 2013 unfortunately skipped my list as I was ignoring "End of Life" products :)

That said, I think I have everything squared as far as L. Please do continue to post your request if you find something missing.

tags07, 2015-09-30:
@darkobas We posted "MMI-LPD23.118-12" for Moto X Play. We'll keep updating the latest release.

socraticbliss, 2015-09-30:
How is it that the Moto X 2013 is considered EOL and yet it is still receiving patches?

If the device was truly EOL, then there wouldn't be any patches released, but I understand that this recent patch was for the Stagefright security vulnerability, so that kernel release I guess is a special case

However, the original request was for the kernel for the 5.1 Lollipop update, which was when the device was not considered EOL yet, but I understand the device priorities, it just is irritating when there is no communication to those that are waiting

tags07, 2015-09-30:
@socraticbliss, The database I use to cross check refers to products no longer manufactured as EOL. Sorry if that caused some confusion.

We will respond to all requests in a timely manner.

socraticbliss, 2015-09-30:
Ahh that now makes total sense, thanks for the clarification :)

Apologies again for my persistence, I just saw other issues were getting responses, yet this one was left unresponded to for quite a long time

Hasuris, 2015-09-30:
I am glad this all works out fine in the end. It's just that I am monitoring this thread for a while now and the lack of progress was quite frustrating. Hence my bit rude reply to this. 

Good to see you guys working on that end. It's the little things like this, that make costumers either very happy or disappointed.

tags07, 2015-09-30:
Kool.. not sure why i missed this thread. Did it have a different subject
or request before?

socraticbliss, 2015-10-02:
Yea, it had it's name changed a few times, it probably didn't help that I closed and re-opened the issue to see if that would increase the chance of viewing it

closed 2015-10-06 by socraticbliss