MotorolaMobilityLLC/kernel-msm #25: Broken headset button (titan, thea)

luk1337, 2015-07-19:
I have samsung headphones with 3 buttons ( vol up, vol dn, headsethook ) and it seems like both thea and titan kernel have some problems with vol dn one since it's detected like once every 3-5 presses. Can somebody confirm that on their devices?
App I used to test these buttons: https://github.com/downloads/chrisboyle/keytest/KeyTest.apk

Alberto97, 2015-07-19:
OnePlus Silver Bullet headphones; headsethook is fine but vol +/- buttons aren't recognized at all on falcon

luk1337, 2015-07-19:
on @gregor160300's headphones vol dn button isn't recognized at all on titan.

gregor160300, 2015-07-20:
Jup can confirm what luk said, my headphones: sennheiser cx 5.00 g work after luk's keylayout mod, but still only volume up and play/pause button, Moto E 2015 has full support for these headset, I tested them on it. Myabe this info makes a fix easier

closed 2015-08-21 by luk1337