MotorolaMobilityLLC/kernel-msm #26: Looking for HID driver for RZ100

hzulla, 2015-07-23:
Hi,

the RZ100 Smart Controller is a bluetooth device sold by Motorola.

It works out of the box on Motorola Android phones, but [not on other manufacturer's Android devices](https://code.google.com/p/android/issues/detail?id=180029). The same is true for Linux desktops.

Obviously, the HID driver needed to use the RZ100 comes pre-installed with Motorola products and nowhere else.

Can the Motorola Mobility team please provide this driver so that it can be included upstream with Linux desktops and stock Android?

Thank you.

awheeler-mmi, 2017-09-28:
As far back as Marshmallow we don't have any specific driver code for the RZ100.  It looks like the device just identifies itself as 2-3 different BT devices, including a full keyboard.

closed 2017-09-28 by awheeler-mmi

hzulla, 2017-10-04:
@awheeler-mmi There appears to be RZ100 specific code needed. I have tried pairing it with a multitude of devices, but the touchpad specific part wasn't supported.

awheeler-mmi, 2017-10-04:
I can't find anyone here that worked on the RZ100.  After some spelunking around I found [this driver](https://github.com/MotorolaMobilityLLC/kernel-msm/blob/kitkat-4.4.4-release-titan/drivers/hid/hid-motorola.c).  This is "the" Motorola HID driver that we used for lapdock and other multitouch devices, but I can't be 100% certain that it supports RZ100. If I can find more info then I'll update this page.

awheeler-mmi, 2017-10-04:
Ah, great!

hzulla, 2017-10-05:
Wow, thanks. I'll try it out.

hzulla, 2018-05-18:
@awheeler-mmi I would very very much love to have the RZ100 support included in mainline. Is there any plan  on your team for adding hid-motorola to mainline?