MotorolaMobilityLLC/kernel-msm #28: CIFS.ko module for XT907

Rocksteady18, 2015-08-13:
Hey guys, I got the msm kernel source and want to build it so i can get this module.  When I try to make it says:

    make[1]: Nothing to be done for `all'.
    make[1]: Nothing to be done for `relocs'.
      CHK     include/linux/version.h
      CHK     include/generated/utsrelease.h
      CALL    scripts/checksyscalls.sh
      CHK     include/generated/compile.h
      SKIPPED include/generated/compile.h
      CC      arch/x86/kernel/apic/io_apic.o
    arch/x86/kernel/apic/io_apic.c:382:13: warning: ‘io_apic_level_ack_pending’ defined but not used     [-Wunused-function]
    error, forbidden warning: io_apic.c:382

I just want cifs on my phone.  Can someone help me?  I'm VMware if that matters.  Thanks!

Rocksteady18, 2015-08-14:
Now it made it but I get this when I insmod it:

insmod: init_module 'cifs.ko' failed (Exec format error)

What to do..

jcarlyle, 2015-08-17:
You appear to be compiling for x86. You should check your cross-compiler setup.

Rocksteady18, 2015-08-19:
Thank you sir, I had the wrong toolchain.  Now I am trying with one I got from android ndk, not sure if it's the right one.  And it still won't compile.  Thanks for the help before but do you know what I am doing wrong this time?

       WARNING: Symbol version dump /media/Max/kernel/Module.symvers
               is missing; modules will have no dependencies and modversions.

      CC [M]  fs/cifs/cifsfs.o
    In file included from /media/Max/kernel/arch/x86/include/asm/bitops.h:505:0,
                 from include/linux/bitops.h:22,
                 from include/linux/kernel.h:19,
                 from include/linux/cache.h:4,
                 from include/linux/time.h:7,
                 from include/linux/stat.h:60,
                 from include/linux/module.h:10,
                 from fs/cifs/cifsfs.c:26:
    /media/Max/kernel/arch/x86/include/asm/arch_hweight.h: In function ‘__arch_hweight64’:
    /media/Max/kernel/arch/x86/include/asm/arch_hweight.h:53:7: error: expected ‘:’ or ‘)’ before     ‘POPCNT64’
    In file included from /media/Max/kernel/arch/x86/include/asm/thread_info.h:22:0,
                 from include/linux/thread_info.h:53,
                 from include/linux/preempt.h:9,
                 from include/linux/spinlock.h:50,
                 from include/linux/seqlock.h:29,
                 from include/linux/time.h:8,
                 from include/linux/stat.h:60,
                 from include/linux/module.h:10,
                 from fs/cifs/cifsfs.c:26:
    /media/Max/kernel/arch/x86/include/asm/processor.h: At top level:
    /media/Max/kernel/arch/x86/include/asm/processor.h:122:30: error:     ‘CONFIG_X86_L1_CACHE_SHIFT’ undeclared here (not in a function)
    /media/Max/kernel/arch/x86/include/asm/processor.h:122:1: error: requested alignment is not an     integer constant
    /media/Max/kernel/arch/x86/include/asm/processor.h:164:0: warning: "cache_line_size" redefined     [enabled by default]
    error, forbidden warning: processor.h:164

luk1337, 2015-08-19:
Try Export ARCH=arm

Rocksteady18, 2015-08-19:
I did, no joy :(

luk1337, 2015-08-19:
Can I ask which branch did you try?

luk1337, 2015-08-19:
also config name

Rocksteady18, 2015-08-19:
branch is falcon-gpe what do you mean by config name?

luk1337, 2015-08-19:
because each device has it's defconfig.

Rocksteady18, 2015-08-19:
how do I check?

luk1337, 2015-08-19:
also falcon-gpe doesn't seem to be dedicated for your device lol

Rocksteady18, 2015-08-19:
Oh dang.  I didn't even realize there are different branches.  Thanks for the help, any idea which branch and how to check config?  Thanks man, very beginnerish at programming here.

luk1337, 2015-08-19:
https://github.com/MotorolaMobilityLLC/kernel-msm/blob/kitkat-4.4.2-release-vanquish/arch/arm/configs/msm8960_defconfig

luk1337, 2015-08-19:
this branch and config

Rocksteady18, 2015-08-19:
thanks, I bet it works with the right ones.  Thanks for the help, I really appreciate it.  How did you find out which branch was for my phone?

luk1337, 2015-08-19:
just do this:
1. Replace "# CONFIG_CIFS is not set" with CONFIG_CIFS=y (arch/arm/configs/msm8960_defconfig)
2. export ARCH=arm
export SUBARCH=arm
export CROSS_COMPILE=/tmp/arm-eabi-4.8/bin/arm-eabi- ( replace with your path to toolchain, I personally use this one https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/arm/arm-eabi-4.8/ )
make msm8960_defconfig
make -jX

luk1337, 2015-08-19:
I just checked the releases tab

luk1337, 2015-08-19:
* or CONFIG_CIFS=m if you want to build it as a module

Rocksteady18, 2015-08-19:
thanks a million!  Can't wait to try, and I do want to build as module.  The whole thing I am trying to do is use the app Mount Manager which needs cifs support.  I have to download the source but then will try again, I really appreciate you and Jeff's help.  I feel like I am learning a lot too.

Rocksteady18, 2015-08-20:
I can't get the toolchain path right, if I do the bin directory, it says permission denied, even though its set to 777 and executable and everything inside.

luca020400, 2015-08-20:
git clone https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/arm/arm-eabi-4.8
sed -i "s/# CONFIG_CIFS is not set/CONFIG_CIFS=m/" arch/arm/configs/msm8960_defconfig
export ARCH=arm
export SUBARCH=arm
export CROSS_COMPILE=`pwd`/arm-eabi-4.8/bin/arm-eabi-
make msm8960_defconfig
make -j5 modules

Rocksteady18, 2015-08-20:
it ends with "no rule to make target 'export'.  Stop"

luca020400, 2015-08-20:
Here it works perfectly
Maybe you cal insom it without any problem --> http://www.mediafire.com/download/nir3hak1rbblfh3/cifs.ko

Rocksteady18, 2015-08-20:
Dang, it said operation not permitted and failed.  I will redo the virtual machine and start from scratch, maybe something in my environment is messed up.   No idea what error 8 is and if it works for you it should work for me.

luca020400, 2015-08-20:
Try to put the module in /system/lib/modules and reboot

Rocksteady18, 2015-08-20:
I did it doesn't show as running in Mount Manager, I don't know how else to check.

Rocksteady18, 2015-08-20:
I need gcc 4.8 for this to work don't I?  My environment has 4.7 and doesn't want to upgrade, is that the problem?

awheeler-mmi, 2017-09-28:
Closing, as this debug session is pretty old.

closed 2017-09-28 by awheeler-mmi