MotorolaMobilityLLC/kernel-msm #31: Kernel Source Request for Moto X Pure Edition (XT1575)

HashBang173, 2015-09-03:
Hi,

Can you please release the Kernel source for the new Moto X Pure Edition (XT1575).

Thanks

charleseb, 2015-09-04:
Hi,

This product is on pre order phase, so we not able to release this version
yet.
Once we are able to release this version I'll inform to you.

Thanks,
Charles

Shawn5162, 2015-09-10:
I see on the Motorola website that the phone is through the pre-order phase, does that mean to source will be released soon? 

Thanks,
Shawn

charleseb, 2015-09-10:
Yes.
Once we are able to release this version I'll start this code push.

Thanks,
Charles

Shawn5162, 2015-09-12:
Any update on the code push. One of my favorite parts of Motorola devices is quick source code releases.

charleseb, 2015-09-15:
Hi Shawn,

We are working to release this version in a few days.

Thanks,
Charles

charleseb, 2015-09-17:
Hi Shawn,

The tag MMI-LPH23.116-18.1 has been released.

Thanks,
Charles

HashBang173, 2015-09-17:
Thank you very much!

closed 2015-09-17 by HashBang173

Shawn5162, 2015-09-17:
Thanks so much