MotorolaMobilityLLC/kernel-msm #36: Build instructions for MMI-LPH23.116-18.1

vadimtk, 2015-10-03:
Can you provide Build instructions for MMI-LPH23.116-18.1 kernel, Moto X Pure Edition ?

tags07, 2015-10-04:
ok will check with the team and post.

ramsudharsan, 2015-10-04:
@tags07 If possible, do update the instructions for all devices here: https://github.com/MotorolaMobilityLLC/readme

tags07, 2015-10-07:
@ramsudharsan Sure will do.

tags07, 2015-10-08:
https://raw.githubusercontent.com/MotorolaMobilityLLC/readme/master/MMI-LPH23.116-18.1.txt   
I made this up looking at the build logs. I have not tested it. Feel free to try it. I'll have someone test it here and fix it if it has bugs..

We'll try to update for other products that have different instructions..if you need something immediate, Let me know..i can prioritize that.

vadimtk, 2015-10-08:
@tags07 great, thanks!

I wonder if you can post command line how to build boot.img after this.
That would be greatly appreciated.

tags07, 2015-10-08:
Well boot.img is out of scope for me.. Long ago I have used mkbootimg to pack it. I am not sure if things have changed. Its been a while I built my own custom boot image. I'll ask around to see if they can post an easy instruction..

May be our friends in XDA can help: http://forum.xda-developers.com/showthread.php?t=2319018

nikhil93uf, 2015-10-08:
So first off, the build instructions seem a bit off. The dtb generated is only for the "bunkerhill" family and none for clark. This means the system won't go past the bootloader. I fixed this by building the dtbs for clark. But now, this brings it back to the same old issue of instant reboot at the unlocked bootloader warning screen

gokart2, 2015-10-22:
Have the "clark" MMI-LPH23.116-18.1 kernel build instructions been created/updated/posted yet? Cannot get it to build, mostly because I have little background on kernel building. Build instructions would be greatly appreciated

vadimtk, 2015-10-22:
@gokart2 build instructions are here
https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-LPH23.116-18.1.txt

and you can check my build script:
https://github.com/vadimtk/moto-x-kernel/blob/master/build-uber.sh

closed 2015-10-22 by tags07