MotorolaMobilityLLC/kernel-msm #37: Motorola razr I, XT890, SMI

HazouPH, 2015-10-06:
Hey there,

Is there any chance the kernel source of the Intel device comes available? We got them all till JB, but are missing the Android 4.4.2 one. And there are to many changes to adapt the JB one to work on the 4.4.2.

Hazou

tags07, 2015-10-07:
@HazouPH , RAZR I kernel sources are posted on sourceforge site: http://sourceforge.net/projects/razr-i.motorola/files/

We don't plan to post the git repos of that product since its an old product. We started to support github for all devices starting Moto X 2013.

If you can let us know the tag/software version/build version, I can check if we still have access to those sources. Its been a while, I am not sure if they are still around.

HazouPH, 2015-10-07:
The RAZR I kernel sources on sourceforge are for 4.1.2. Around April this year we got the 4.4.2 update and that uses a slightly different kernel modules with major updates to some modules where we don't have access to.

Some info:
ro.build.id=KIA20.74
ro.build.version.release=4.4.2
ro.board.platform=mrst
ro.build.product=smi
ro.build.fingerprint=motorola/XT890_rteu/smi:4.4.2/KIA20.74/55:user/release-keys
ro.build.version.full=Blur_Version.990.43.74.XT890.Retail.en.EU
Kernel version 3.0.34+ build on 14 aug 2014

If u have even more sources than can be made available, I would happily look into them on sourceforge

HazouPH, 2015-10-26:
Is there maybe any news/update on this question?

tags07, 2015-10-27:
Its in the pipeline. We'll post them soon.

lelapincretin8, 2015-11-03:
Any news about the release ? Will it be on GitHub or SourceForge?

charleseb, 2015-11-03:
Hi lelapincretin8,

I'm working to release this version, it will be available in a few days.

Thanks,
Charles

lelapincretin8, 2015-11-03:
Thanks for your fast reply :)

lelapincretin8, 2015-11-10:
Hi Charles,
Thank you for the source code release. Can we get a Readme file for all the building instructions on Sourceforge please?

HazouPH, 2015-11-10:
I am glad U released some sources. But if I compare them with the 4.1.2 sources, there are almost no differences and if I build them (kernel) It still will fail with the exact same problem. The kernel sources is missing updates. Especially in the driver/staging/mrst (version 1.9 to 1.12) and it is missing the modem_control driver, maybe a lot mroe. Also the defconfig doesn include the audit, sepolicy rules.

Did u maybe made a mistake and copied the old kernel sources as the new ones?

Thanks for all the work though!!

tags07, 2015-11-10:
@HazouPH What are you using for defconfig?

( perl -le 'print "# This file was automatically generated from:\n#\t" . join("\n#\t", @ARGV) . "\n"' hardware/intel/linux-2.6/arch/x86/configs/i386_mfld_moto_defconfig && cat hardware/intel/linux-2.6/arch/x86/configs/i386_mfld_moto_defconfig ) > .config

also, what do you mean by modem_control driver? If you know the names of files you think are missing we can check. btw, Charles is checking the version he posted to be sure the version is correct. We'll let you know if there was an error.

HazouPH, 2015-11-11:
To test everything I am using the stock defconfig from Moto: i386_mfld_moto_defconfig. Found in the linux-2.6 package in the latest release (a day ago).

This defconfig already missed some audit/sepolicy rules which on android 4.4.2 are needed and some more. The stock kernel has full selinux support. But the gernerated one doesn't. So I already was sceptical if this would work on the stock rom (4.4.2).

I know for sure the following files are missing and/or are changed in the new kernel:
- drivers/modem_control/*
- drivers/staging/mrst/*
- drivers/staging/intel_media/*

Most likely changed:
- sound/soc/intel-mid/*
- arch/x86/platform/intel-mid/*

By extracting the strings from the kernel, I found many of the following strings:
/localrepo/hudson/workspace/KIA20.X-PD-XT890-RETAIL-BR-EDF-KK-signed-cid/platform/hardware/intel/linux-2.6/drivers/staging/intel_media/bld/clovertrail/sgx/../../../graphics/sgx/services4/srvkm/env/linux/mm.c

Maybe it helps at locating the right source? This is the BR version, but I don think there will be much differences from other regions in kernel source.

Thanks!

tags07, 2015-11-11:
@HazouPH, I just downloaded what was posted by Charles and compared it against the version i have checkout. Indeed they are different. Sorry about that. We will repost the correct version. 

thanks,
Gopi

HazouPH, 2015-11-11:
Thank u very much for correcting that. Will see them coming.

Are u only updating the kernel source or also adding\replacing other packages?

charleseb, 2015-11-11:
Hi HazouPH,

I'm working to conclude this task and I'll replace all packages in a few
hours.

Tks,
Charles

charleseb, 2015-11-11:
Hi HazouPH,

I just uploaded the correct packages. I think we are good now.
Sorry about that.

Thanks,
Charles

HazouPH, 2015-11-12:
Hi Charleseb and tags07,

It's all looking fine. The kernel works when compiled and is not missing any file as far as I can see.

Thanks for all your effort!!

HazouPH

closed 2015-11-12 by HazouPH

HazouPH, 2015-12-15:
I have one more question and I hope u have an answer for it.

The kernel works great, but I did some stupid mistake with kexec, and the device won't boot again. There is no access to AP fastboot modus or android or any other methode where fastboot and adb can be used. It keeps sitting on the very first bootlogo and never goes any further. I have the feeling the bootloader is corrupt.

Now on x86 devices we have a so called "SoC" mode or "DnX" modes. I can access that and flash the firmware files. Here is the problem. 1 File has never been provided. Is there a chance u can provide me with the following files from the latest release, I don't know the exact name, but they should be the files:

- IFWI.bin
- FW_DNX.bin
- OS_DNX.bin
- Droidboot.img

And if I am correct they are found within the map:
vendor/intel/private/ifwi or something like that.

This may also repair some other problems we have encountered in the Razr I community.

Thanks!