MotorolaMobilityLLC/kernel-msm #38: Booting Moto X pure with modified stock kernel reboots at unlocked bootloader screen

nikhil93uf, 2015-10-07:
Due to the lack of any build instructions, I have approached the build in the general sense. Compiling the kernel and packing it with the stock untouched RAMDisk and a generated DT image causes instant reboot at the unlocked bootloader warning screen. I'm not sure what needs to be done here since this method worked fine on other phones I have compiled kernels for stock.

tags07, 2015-10-07:
We are working on the kernel build instructions. Will post them soon. Packaging is out of scope but you should be able to figure that.

vadimtk, 2015-10-08:
@nikhil93uf can you share instructions how you build boot.img?

nikhil93uf, 2015-10-08:
@vadimtk I simply reused the parameters generated by the unpack tool to repack with only the dt image and the zImage modified.

vadimtk, 2015-10-08:
@nikhil93uf ok, so I am also getting boot-loop, so I am trying to figure out if this I am doing something wrong or dt image is not built properly.

nikhil93uf, 2015-10-08:
The dt image should be OK (though I am not very confident). I'm assuming the modified kernel is failing some sort of a verification process.

jhrycay, 2015-10-09:
I don't know which Moto X you're referring to but check the original DT you extracted. It may be pre-pended with a "lookup" table:

$ hexdump -C -n 1024 dt.img
00000000  51 43 44 54 03 00 00 00  02 00 00 00 f6 00 00 00  |QCDT............|
00000010  41 00 00 00 00 81 00 00  01 00 02 00 09 00 02 00  |A...............|
....

If so, take a look at CAF:

https://www.codeaurora.org/cgit/quic/la/device/qcom/common/tree/dtbtool?id=a0405b8ab7ffae9764ac5a9e67d53cefa9719fb1

on how the DTB portion of the boot.img gets populated.

nikhil93uf, 2015-10-12:
Well, I ran the hexdump and the output is similar to what you described above. I am very positive it's SELinux preventing me from booting the kernel. I could replicate the issue on a non-stock firmware with SELinux enabled. It booted fine with SELinux disabled. Forcing it to work in permissive mode made no difference either. 

Here's the output of the hexdump for your reference
http://pastebin.com/rhV7FkJ8

jhrycay, 2015-10-14:
Yes, your DTB looks fine. I was able to get a device booting here (to be fair its engineering unit). But simply re-using the extracted dt and ramdisk, and using the mkbootimg from our system/core git, I used the following mkbootimg:

> $MKBOOTIMG --kernel out/target/product/generic/obj/kernel/arch/arm64/boot/Image --ramdisk ./original_ramdisk.gz --cmdline "console=ttyHSL0,115200,n8 androidboot.console=ttyHSL0 androidboot.hardware=qcom msm_rtb.filter=0x37 ehci-hcd.park=3 lpm_levels.sleep_disabled=1 earlyprintk=msm_hsl_uart,0xf991e000 utags.blkdev=/dev/block/bootdevice/by-name/utags utags.backup=/dev/block/bootdevice/by-name/utagsBackup boot_cpus=0-3" --base 0x00000000 --pagesize 4096 --dt ./original_devtree.img  --output ./new_boot.img

Note the uncompressed --kernel parameter

vadimtk, 2015-10-14:
@jhrycay thanks! just to make sure, how do you get original_devtree.img ?

jhrycay, 2015-10-14:
If you checkout:

https://github.com/MotorolaMobilityLLC/system-core/blob/kitkat-mr1-release-falcon-gpe/mkbootimg/bootimg.h#L43

you can see the boot.img header has the DTB size packed into a previously 'unused' field. You could find some boot.img extractors like:

https://gist.github.com/Evisceration/84c7ebc64f126bd6b3a8

It looks like it supports the QC boot.img header format. If you are making changes to the DTBs you'd have to use the dtbtool I sent a link on CAF

savoca, 2015-10-14:
@vadimtk You can build dtb images yourself for most QC devices if you use CyanogenMod's fork of dtbTool here: https://github.com/CyanogenMod/android_device_qcom_common/tree/cm-12.1/dtbtool

Creating dt.img using clark source from Moto:
http://paste.ubuntu.com/12783325/

nikhil93uf, 2015-10-14:
@savoca @jhrycay I used the same exact procedure as you described for both the mkbootimg and the dt image. It just won't go past the warning screen. As I described previously, experimenting with SELinux shows that disabling it allows me to boot, but then of course the system denies permissions to the framebuffer and thus it remains stuck at the bootlogo. Should I be generating some sort of context to allow my kernel to be accepted?

jhrycay, 2015-10-14:
@nikhil93uf want to share the boot.img?

nikhil93uf, 2015-10-14:
@jhrycay Sure.

https://drive.google.com/file/d/0B_UwKO6VqkKgam1mVVhaVkVUMnc/view?usp=sharing

jhrycay, 2015-10-15:
Definitely crashes here too. Its triggering a watchdog or TZ/RPM reset. I see a lot of 'SCM' call failures and LPASS failing to get out of reset. What local changes do you have? I see the top commit of ours are the same but yours is dirty. 

my gcc: gcc version 4.9.x-google 20140827 (prerelease) (GCC) 
yours: gcc version 4.9.x-google 20150123 (prerelease)

can't imagine that would have an affect though, but upstream tag of "android-5.0.0_r5.1" is what I used. 

Also, what is your f2fs top commit? I see some f2fs errors forcing recovery:

7b8c767 f2fs: skip committing valid superblock

I repackaged my kernel image with your DT and RAMDisk and it booted again, so something looks a little different on the kernel image. I wasn't able to extract the ikconfig from your image either, can you share your .config I'm wondering if its the wrong one?

nikhil93uf, 2015-10-15:
Here's my config: http://pastebin.com/Wv7uTsWx

And my f2fs top commit is exactly the same as yours
https://github.com/MotorolaMobilityLLC/motorola-kernel/commit/7b8c767615d28d2ae0a441d0eeeb86d252209989

nikhil93uf, 2015-10-15:
So I extracted the config from the stock image and seems like there's something fishy with the modules. Maybe that's causing the resets? @jhrycay

jhrycay, 2015-10-15:
The config I built off of msm8994-perf_defconfig doesn't match up with yours, and modules definitely are off on configuration. Did you modify the config?

> CONFIG_MODULES=y
> \# CONFIG_MODULE_FORCE_LOAD is not set
> CONFIG_MODULE_UNLOAD=y
> CONFIG_MODULE_FORCE_UNLOAD=y
> CONFIG_MODVERSIONS=y
> \# CONFIG_MODULE_SRCVERSION_ALL is not set
> \# CONFIG_MODULE_SIG is not set
> CONFIG_MODULE_WHITELIST=y
> CONFIG_MODULE_EXTRA_COPY=y

versus your:

> \# CONFIG_MODULES is not set

nikhil93uf, 2015-10-15:
Well I am not sure how that creeped in. In any case, I tried it with on and I still have the same issue.

Here's my updated config: http://pastebin.com/itDghs41
And here's the corresponding boot image: https://drive.google.com/file/d/0B_UwKO6VqkKgYnllam9GNlhoOE0/view?usp=sharing

This is the exact same config as the stock config I extracted using ikconfig

jhrycay, 2015-10-15:
Still crashes here and confirm ikconfig matches now.

I then tried a shot in the dark and checked out:

94cc923 [linux-x86] Refresh gcc 4.9 prebuilts for aarch64

to get to your gcc and am crashing as well. So, if you want to try your side checking out:

d9735fc [linux-x86] ARM toolchain refresh

I'm wondering if it has to do with the kernel arm errata work arounds that were put in before GCC toolchain was fixed?

jhrycay, 2015-10-15:
@nikhil93uf make sure to wipe the out kernel directory though

nikhil93uf, 2015-10-15:
But doesn't the d9735fc [linux-x86] ARM toolchain refresh refer to a 32bit toolchain?

jhrycay, 2015-10-15:
@nikhil93uf my bad, I was in the 4.8 directory:

e1a8a48 [linux-x86] Refresh gcc prebuilts for aarch64 Refresh gcc-4.9 prebuilts for aarch64. This new prebuilts contain (1) workaround for cortex-a53 erratum 843419, (2) some compiler b

matches my 20140827 datecode.

nikhil93uf, 2015-10-15:
@jhrycay 
Alright, switched it out for that toolchain. Now it just gets stuck at the warning screen. Doesn't reboot anymore though. I also noticed the MTP is recognized (albeit with nothing mounted) on the PC. Seems like it boots but nothing's actually happening on the screen.

Here's the image: https://drive.google.com/file/d/0B_UwKO6VqkKgQml0VUJNWDRJTDA/view?usp=sharing

nikhil93uf, 2015-10-15:
@jhrycay Ignore my previous comment. It boots! Thank you! So it was the toolchain. Pretty weird since it seems to boot fine on a non-stock ROM compiled using the same toolchain.

jhrycay, 2015-10-15:
@nikhil93uf was just typing that it was booting here too. I'm not sure what's up with the toolchain / kernel combination here. Curious as to what ROM was booting okay for you? Perhaps its on a later upstream CAF.

nikhil93uf, 2015-10-15:
You're right. The manifest for the ROM (CM) I built pulls in the toolchain from a custom repo and not from Google. I did not notice that. Funny how the most recent commit for both repos lines up though.

jhrycay, 2015-10-15:
Yea, strange that it was the toolchain. Enjoy. One thing you'll probably want to disable or update the ramdisk for is the CONFIG_MODULE_WHITELIST if you see:

[   30.842373,1] module-whitelist: Module hash not found in whitelist: bb4b337aabed2b626c666148e12a73953b59e93b

which I do with your image, but I just might have a different system.img

nikhil93uf, 2015-10-15:
@jhrycay I don't see that. I replaced the modules too. Interesting it reports the hashes still match.

awheeler-mmi, 2017-09-28:
Thanks, Jason!

closed 2017-09-28 by awheeler-mmi