MotorolaMobilityLLC/kernel-msm #39: Missing fs/f2fs directory in lollipop-5.1.1-release-ghost (and others)

jesultra, 2015-10-07:
The file fs/Kconfig in the source code for lollipop-5.1.1-release-ghost refers to a directory fs/f2fs, but that directory is not present in the .zip or .tar.gz files, nor browsing git repository.

Please upload the missing files to github, as I believe that the f2fs filesystem is indeed used for the /data filesystem on first generation Moto X devices (such as 1053).

dhacker29, 2015-10-07:
The files are contained here https://github.com/MotorolaMobilityLLC/motorola-kernel and symlinked (see the readme) but there is no matching branch for ghost release.

tags07, 2015-10-07:
oops did we miss the tag/branch for ghost? Can you send me the specific tag/branch? I'll see if we need to push. If the commits are already in, its possible we forgot to apply a tag.

jesultra, 2015-10-07:
I am not familiar with GIT terms.  I can say that when I look in this directory, I would have expected to see a f2fs subdirectory:
https://github.com/MotorolaMobilityLLC/kernel-msm/tree/lollipop-5.1.1-release-ghost/fs

I think that the same is true for the other flavors of lollipop-5.1.1-release. (I looked in one other, lollipop-5.1.1-release-osprey, where it was also absent.)

dhacker29, 2015-10-08:
What I was trying to say earlier is that Motorola maintains one copy of the fs/f2fs subdirectory in https://github.com/MotorolaMobilityLLC/motorola-kernel you can either clone that and symlink the files in or add them to the kernel tree.  There is no lollipop-5.1.1-release branch there so you will have to look up the tag of the build you want.

gokart2, 2015-10-09:
Believe it is labeled wrong.....meaning its a f2fs file instead of an f2fs.c file? Regardless, it won't compile cause it can't open that specific file.

tags07, 2015-10-09:
@gokart2 can you explain what you mean? The dependency is on Kconfig... what do you mean by f2fs.c 
What is not compiling?

ccfries, 2015-10-09:
that f2fs "file" is a symlink to motorola/kernel .. get the separate
repository for f2fs sources.  (We hacked that to avoid propagation
headaches across many kernel branches, sorry for the confusion)

gokart2, 2015-10-09:
I apologize for my confusing post. This is my first kernel rodeo.

I'm not in front of my Linux PC to give specific the specific error, but it has an fs/f2fs/kconfig error. Basically the fs/f2fs file isn't there. I used the f2fs file from the link DHacker posted, and it doesn't error now.

gokart2, 2015-10-09:
This file is what I used to make it stop the error.
https://github.com/MotorolaMobilityLLC/motorola-kernel/tree/lollipop-5.0.2-release/fs/f2fs

jesultra, 2015-10-09:
Thank you dhacker29 and tags07 for explaining the issue to the noobs.
I see that the motorola-kernel seems to have all the same tags as kernel-msm, so it is just a matter of selecting the same. (Or going with a branch of either or both, if you want post-release improvements in your custom kernel build.)

tags07, 2015-10-09:
I think it might help if we just add a line in the build instructions to symlink to the appropriate folder. That might clarify things a bit too.

closed 2017-09-27 by awheeler-mmi