MotorolaMobilityLLC/kernel-msm #40: Missing kernel patch to address sound distortions during deep-buffer playback

myzb, 2015-10-13:
Present since lollipop on all the msm8226. Cyanogenmod spotted the missing patch from CAF here:
https://github.com/CyanogenMod/android_kernel_motorola_msm8226/commit/f67ddf18eec92729c88cadfdef65fdd9775126f1

It may be worthwhile to check if this also addresses the issue on the stock Motorola kernel for the Moto G msm8226 lineup. The the problem was caused by changes in 'platform.h' within the audio HAL. 

_platform.h_
```
@@ -157,7 +157,7 @@ enum {
  * the buffer size of an input/output stream
  */
 #define DEEP_BUFFER_OUTPUT_PERIOD_SIZE 960
-#define DEEP_BUFFER_OUTPUT_PERIOD_COUNT 4
+#define DEEP_BUFFER_OUTPUT_PERIOD_COUNT 5
 #define LOW_LATENCY_OUTPUT_PERIOD_SIZE 240
 #define LOW_LATENCY_OUTPUT_PERIOD_COUNT 2
```

myzb, 2016-04-10:
Guess nobody cared ...

closed 2016-04-10 by myzb

luca020400, 2016-04-10:
@tags07 @charleseb ^^

luca020400, 2016-04-10:
For reference this is the device side change https://github.com/CyanogenMod/android_device_motorola_msm8226-common/commit/563391841224d0622199e3bf0b08ca0bf1d3ef32