MotorolaMobilityLLC/kernel-msm #44: Missing kernel source for Condor Lollipop 5.1.1

shanmukhateja, 2015-11-14:
As the title suggests, there is no branch for kernel source for Moto E (1st Gen) Condor for the Lollipop 5.1.1 ROM. All there is for Condor is the Kitkat source which I highly doubt will boot considering the changes between the ROMs

tags07, 2015-11-17:
What build version are you looking for?

shanmukhateja, 2015-11-17:
The system version is: 23.26.8.condor_retaildsds.retaildsdsall.en.03.reteu

The Build number is: LPC23.13-34.8

tags07, 2015-11-17:
Ok thanks.. that is in our pipeline. We'll post them soon.

charleseb, 2015-11-19:
Hi shanmukhateja,

The version LPC23.13-34.8 has been released to Github.

Thanks,
Charles

closed 2015-11-19 by charleseb

MLXProjects, 2020-05-19:
Unzipped the source file (kernel-msm-MMI-LPC23.13-34.8.zip), and while trying to compile it complains about f2fs folder being missing (checking the zip shows f2fs as a 29-bytes file). I can just edit the files removing the references to it, but releasing a uncompilable source (without mods) isn't the correct way of doing things. I've tried downloading the file in two PCs with different ISPs just for checking if the download was corrupted. which isn't the case. I may be wrong, but as far as i've tried, the source is impossible to build without modding it.

MLXProjects, 2020-05-19:
Unzipped the source file (kernel-msm-MMI-LPC23.13-34.8.zip), and while trying to compile it complains about f2fs folder being missing (it points to a inexistent folder, ../../motorola/kernel/fs/f2fs which doesn't exist on the zip). I can just edit the files removing the references to it, but releasing a uncompilable source (without mods) isn't the correct way of doing things. I've tried downloading the file in two PCs with different ISPs just for checking if the download was corrupted. which isn't the case. I may be wrong, but as far as i've tried, the source is impossible to build without modding it.

awheeler-mmi, 2020-05-19:
Hey @MLXProjects ,

motorola/kernel is a separate project.  It is [here](https://github.com/MotorolaMobilityLLC/motorola-kernel/tree/MMI-LPC23.13-34.8).  This should be described in the readme.

MLXProjects, 2020-05-19:
That's not described neither in the ernel-msm's readme nor in the source zip (maybe I'm looking at the wrong place?). 
Anyway, thanks for the fast answer! That was impressive.

rajatgupta1998, 2020-05-19:
> That's not described neither in the kernel-msm's readme nor in the source zip (maybe I'm looking at the wrong place?).
> Anyway, thanks for the fast answer! That was impressive.

https://github.com/MotorolaMobilityLLC/readme

Might check this MMI-LPH23.116-18.1.txt or something similar for ur device.

awheeler-mmi, 2020-05-19:
Awesome, great to hear it's working.