MotorolaMobilityLLC/kernel-msm #45: victara 6.0 release source

invisiblek, 2015-11-17:


luca020400, 2015-11-29:
We're still waiting

invisiblek, 2015-11-29:
you guys got beat by HTC to release kernel source

that's pretty pathetic

tags07, 2015-11-29:
Closing.. No description!!!

closed 2015-11-29 by tags07

invisiblek, 2015-11-29:
Are you serious? i'll give you a fucking description.

This shit is GPL, you guys released a build and no source is available. When asked for source you're going to ignore the request?

Here's the ROM build: MPE24.49-18
Here's the kernel string: Linux localhost 3.4.42-ga9982b3-00002-g389e46b #1 SMP PREEMPT Tue Nov 3 16:12:18 CST 2015 armv7l

invisiblek, 2015-11-29:
https://github.com/MotorolaMobilityLLC/kernel-msm/issues/48

tags07, 2015-11-29:
@invisiblek : Dude, in your original post, you never requested the source. There was no description. The title "victara 6.0 release source" was not clear to me if you are commenting on something or asking for the source. Please be clear on what you want and you'll get a decent response. What do you think we have a mind reading machine here that interprets your cryptic messages?

u-ra, 2015-11-29:
That's a nice rant, but you forgot to acknowledge you got beat by *HTC* to release the kernel source. That's pretty pathetic.

kumajaya, 2015-11-30:
LOL :D

charleseb, 2015-12-29:
Hi invisiblek,

The tag related to the version MMI-MPE24.49-18 has been released.

Thanks,
Charles