MotorolaMobilityLLC/kernel-msm #48: victara 6.0 source missing

invisiblek, 2015-11-29:
Here's the ROM build: MPE24.49-18
Here's the kernel string: Linux localhost 3.4.42-ga9982b3-00002-g389e46b #1 SMP PREEMPT Tue Nov 3 16:12:18 CST 2015 armv7l

tags07, 2015-11-29:
Thank you for the request. We'll check if we shipped devices with the above version and post the kernel sources soon. Its a holiday week here in US. So please be patient. Its in the pipeline.

tags07, 2015-12-16:
The following are some of the tags in the pipeline..Some of them may have been posted already:
We'll try to prioritize M releases first. Thanks for your patience while we get these going...

KXB21.85-23
LCD23.118-21.2
LCK23.130-23
LPA23.12-39.8
LPB23.13-17.3
LPB23.13-61
LPD23.118-32
LPD23.215
LPE23.32-52.3
LPH23.116-18.1
LPH23.145-7
LPI23.29-22-R.10
LPI23.72-66
LPK23.145-5
LXA22.54-17
LXB22.99-24.13
LXI22.50-62.2
MPE24.49-18
MPH24.49-18
MPI24.65-24
SU6-7.3

charleseb, 2015-12-29:
Hi invisiblek,

The tag related to the version MMI-MPE24.49-18 has been released.

Thanks,
Charles

closed 2015-12-29 by charleseb