MotorolaMobilityLLC/kernel-msm #49: GPL source request for clark (Moto X Style/Pure Edition)

HashBang173, 2015-11-29:
Hi,

There is currently no source posted for Android 6.0 for clark

Build number: MPH24.49-18

tags07, 2015-12-02:
We are working on all Android M sources. Waiting for the SW to rollout to a few more countries/careers so we can all the commits the the most recent deployment. Thanks for being patient :)

ramsudharsan, 2015-12-03:
@tags07 :+1:

randomblame, 2015-12-04:
this is me being patient :D

Shawn5162, 2015-12-06:
@tags07 Waiting eagerly

vadimtk, 2015-12-15:
@tags07 :+1:

tags07, 2015-12-16:
The following are some of the tags in the pipeline..Some of them may have been posted already:
We'll try to prioritize M releases first. Thanks for your patience while we get these going...

KXB21.85-23
LCD23.118-21.2
LCK23.130-23
LPA23.12-39.8
LPB23.13-17.3
LPB23.13-61
LPD23.118-32
LPD23.215
LPE23.32-52.3
LPH23.116-18.1
LPH23.145-7
LPI23.29-22-R.10
LPI23.72-66
LPK23.145-5
LXA22.54-17
LXB22.99-24.13
LXI22.50-62.2
MPE24.49-18
MPH24.49-18
MPI24.65-24
SU6-7.3

pattuspl, 2015-12-19:
+1 please post as soon as possible :)

charleseb, 2015-12-23:
Hi all,

The version tag MMI-MPH24.49-18 has been released.

Thanks,
Charles

closed 2015-12-23 by charleseb