MotorolaMobilityLLC/kernel-msm #54: defconfig for osprey (Moto G 3rd Gen)

lightydo, 2015-12-16:
which defconfig should be used to build the kernel for Moto G 3rd Gen (osprey) ?

tags07, 2015-12-29:
Can you post the build version, system version, so i know what version of osprey you are looking for?

lightydo, 2015-12-30:
I don't have the device yet. I will post as soon as I have it. 
Thanks.

lightydo, 2016-02-08:
here is the info: 

[ro.build.description]: [osprey_reteu-user 5.1.1 LPI23.72-47 50 release-keys]
[ro.build.display.id]: [LPI23.72-47]
[ro.build.flavor]: [osprey_reteu-user]
[ro.build.host]: [ilclbld35]
[ro.build.id]: [LPI23.72-47]
[ro.build.product]: [osprey_umts]
[ro.build.tags]: [release-keys]
[ro.build.type]: [user]
[ro.build.user]: [hudsoncm]
[ro.build.version.all_codenames]: [REL]
[ro.build.version.codename]: [REL]
[ro.build.version.full]: [Blur_Version.23.41.47.osprey_reteu.reteu.en.EU]
[ro.build.version.incremental]: [50]
[ro.build.version.qcom]: [AU_LINUX_ANDROID_LA.BR.1.1.3_RB1.05.01.00.032.015]
[ro.build.version.release]: [5.1.1]
[ro.build.version.sdk]: [22]

lightydo, 2016-02-09:
Please note, I succefully built the kernel with msm8916_defconfig and booted the device with it. 
The only problem I have now is that wlan won;t work. I pulled the ko file and signed it again with the my kernel keys but now it complains about:
insmod: init_module '/system/lib/modules/pronto/pronto_wlan.ko' failed (Exec format error)

dmesg shows: 
 <4>[ 2035.892969,0] QSEECOM: qseecom_load_app: App (keymaster) does'nt exist, loading apps for first time
<3>[ 2035.910311,0] scm_call failed: func id 0x32000101, arginfo: 0x3, args: 0x1a5c, 0x837c, 0x8ec30000, 0x0, ret: -2, syscall returns: 0xffffffda, 0x0, 0x0
<3>[ 2035.910323,0] QSEECOM: qseecom_load_app: scm_call to load app failed
<3>[ 2035.910354,0] QSEECOM: qseecom_ioctl: failed load_app request: -22
<4>[ 2037.701394,0] wlan: disagrees about version of symbol module_layout

any idea what to do?

mayurdjain, 2016-02-10:
What tool chain did you use to build MOTO G 3rd Gen XT1550 ?

lightydo, 2016-02-10:
arm-eabi-4.7, taken from google
https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/arm/

mayurdjain, 2016-02-10:
As per my understanding, for MOTO G 3rd Gen (with Marshmellow update) you have to use below kernel 
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-MPI24.65-24
Please check the build number under setting->About phone. If it is MPI 24.65.-25 then you have to use above kernel. 
what are the steps you have used to build kernel? Is it Lollipop or Marshmellow?

lightydo, 2016-02-10:
it is lollipop. 
//[ro.boot.hardware.sku]: [XT1541]
//[ro.build.description]: [osprey_reteu-user 5.1.1 LPI23.72-47 50 release-keys]
//[ro.build.display.id]: [LPI23.72-47]
//[ro.build.version.release]: [5.1.1]

lightydo, 2016-02-10:
mayurdjain, are you affiliated with Motorola LLC or just following this post?

tags07, 2016-02-10:
@lightydo : Did you pick the kernel and wlan driver from here:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-LPI23.72-47.5
https://github.com/MotorolaMobilityLLC/vendor-qcom-opensource-wlan-prima/releases/tag/MMI-LPI23.72-47.5 ?

Did you use the following defconfig to build kernel: kernel/arch/arm/configs/msm8916-perf_defconfig

Did you build the wlan driver something like this:
`make -C kernel M=../vendor/qcom/opensource/wlan/prima O=$KERNEL_OBJ ARCH=arm CROSS_COMPILE=arm-eabi-  modules WLAN_ROOT=../vendor/qcom/opensource/wlan/prima MODNAME=wlan BOARD_PLATFORM=msm8916 CONFIG_PRONTO_WLAN=m`

Try these and let us know if you still see the errors.

mayurdjain, 2016-02-11:
LIghtydo, I am not affiliated with Motorola LLC. I am following this thread because I am also working on MOTO G 3rd Gen kernel.

lightydo, 2016-02-11:
@tags07 Thanks for the info. Will try it out. 
@mayurdjain here is the process: (I am giving you the details here but I leave it to you as an exercise to put it all into bash scripts to automate the process....)

1. Unlock the Moto G bootloader (you can follow the instructions in installing TWRP and rooting the device here: http://theunlockr.com/2015/09/09/how-to-root-the-motorola-moto-g-3rd-gen/ )
2. Once you have TWRP you can use dd to copy the boot partition and save it (very important so you can always put it back!) -- "dd if=/dev/block/platform/soc.0/by-name/boot of=/sdcard/MotoG.orig.boot.img"
3. Compile the kernel, i.e.: 
export CONFIG_FILE="msm8916_defconfig"
export ARCH="arm"
export CROSS_COMPILE="arm-eabi-"
export TOOL_CHAIN_PATH="$HOME/lucid/tools/arm-eabi-4.8/bin"
export CONFIG_ABS_PATH="arch/${ARCH}/configs/${CONFIG_FILE}"
export PATH=$PATH:${TOOL_CHAIN_PATH}
make -j16 $CONFIG_FILE
make -j16 
follow @tags07 instructions above for the WLAN module. 

4. Once you have the zImage ready, you will need the MotoG.orig.boot.img to disassemble it using unpackbootimg and repack it with mkbootimg. you can get the tools here (I don't remember where I took mine from): https://github.com/osm0sis/mkbootimg 
Then:
./unpackbootimg -i MotoG3Gen_Boot_Orig.img -o out
cp ../arch/arm/boot/zImage out/MotoG3Gen_Boot_Orig.img-zImage
mkbootimg --kernel "$OUT_DIR/MotoG3Gen_Boot_Orig.img"-zImage \
  --ramdisk "$OUT_DIR/MotoG3Gen_Boot_Orig.img"-ramdisk.* \
  --dt "$OUT_DIR/MotoG3Gen_Boot_Orig.img"-dtb \
  --cmdline "$(cat "$OUT_DIR/MotoG3Gen_Boot_Orig.img"-cmdline)" \
  --pagesize "$(cat "$OUT_DIR/MotoG3Gen_Boot_Orig.img"-pagesize)" \
  --base "$(cat "$OUT_DIR/MotoG3Gen_Boot_Orig.img"-base)" \
  --ramdisk_offset "$(cat "$OUT_DIR/MotoG3Gen_Boot_Orig.img"-ramdiskoff)" \
  --tags_offset "$(cat "$OUT_DIR/MotoG3Gen_Boot_Orig.img"-tagsoff)" \
  --output "newboot.img"

This will create "newboot.img" which you can than boot or flash using fastboot:
adb reboot bootloader
fastboot boot newboot.img
or 
fasboot flash boot newboot.img

you will need to copy the new wlan module into your /system/lib/modules...

Hope this is detailed enough. Let me know if there is anything I can help you with. 

Daniel.

mayurdjain, 2016-02-17:
I took kernel source code from https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-LPI23.72-47.5 and tried to build using below steps. 

export CONFIG_FILE="msm8916_defconfig"
export ARCH="arm"
export CROSS_COMPILE="arm-eabi-"
export TOOL_CHAIN_PATH="${HOME}/sources/toolchain/arm-eabi-4.8/bin"
export CONFIG_ABS_PATH="arch/${ARCH}/configs/${CONFIG_FILE}"
export PATH=$PATH:${TOOL_CHAIN_PATH}
make -j16 $CONFIG_FILE

It does not build, I got below error 

$ make -j16 $CONFIG_FILE
kernel/irq/Kconfig:22:warning: leading whitespace ignored
fs/Kconfig:219: can't open file "fs/f2fs/Kconfig"
make[1]: *** [msm8916_defconfig] Error 1
make: *** [msm8916_defconfig] Error 2

Similarly, If I build android M code obtained from 
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-MPI24.65-24
I got below error 

make -j16 $CONFIG_FILE
kernel/irq/Kconfig:22:warning: leading whitespace ignored
drivers/usb/misc/Kconfig:160: can't open file "drivers/usb/misc/sisusbvga/Kconfig"
make[1]: *** [msm8916_defconfig] Error 1
make: *** [msm8916_defconfig] Error 2

Did I miss any step or using a wrong toolchain? 

Thank You
-Mayur

lightydo, 2016-02-17:
try msm8916-perf_defconfig

lightydo, 2016-02-17:
also try: make V=1 $CONFIG_FILE to get more output...

mayurdjain, 2016-02-17:
there is no f2fs folder under fs directory. Did you copy this folder externally? 

Output of make V=1 $CONFIG_FILE is given below 

make V=1 $CONFIG_FILE
make -f scripts/Makefile.build obj=scripts/basic
rm -f .tmp_quiet_recordmcount
mkdir -p include/linux include/config
make -f scripts/Makefile.build obj=scripts/kconfig msm8916_defconfig
scripts/kconfig/conf --defconfig=arch/arm/configs/msm8916_defconfig Kconfig
kernel/irq/Kconfig:22:warning: leading whitespace ignored
fs/Kconfig:219: can't open file "fs/f2fs/Kconfig"
make[1]: *** [msm8916_defconfig] Error 1
make: *** [msm8916_defconfig] Error 2

As there is no f2fs folder it's failing as per my understanding. Please check your sources and confirm. 
Thank You!
-Mayur

mayurdjain, 2016-02-17:
I have tried with msm8916_defconfig and msm8916-perf_defconfig and got the same failure.

mayurdjain, 2016-02-17:
I got the root cause of the build failure. It's due to F2FS folder missing in kernel/FS/
I have commented following line related to f2fs in FS/Kconfig and source code builds fine with msm8916-perf_defconfig and msm8916_defconfig. I am able to build MMI-MPI24.65-24 and MMI-LPI23.72-47.5 with arm-eabi-4.7 toolchain. 
#source "fs/f2fs/Kconfig"

Please check and let me know if you have f2fs folder under your fs tree
Thank You!
-Mayur

lightydo, 2016-02-17:
yes, I do have an fs/f2fs folder...
it is a simlink to the motroloa kernel folder. (fs/f2fs -> ../../motorola/kernel/fs/f2fs/)

you should have: 
motorola/ (inside there is a simlink to motorola-kernel, i.e. kernel -> ../motorola-kernel/)
motorola-kernel/
readme/
kernel-msm/

lightydo, 2016-02-17:
get these: 
https://github.com/MotorolaMobilityLLC/motorola-kernel
https://github.com/MotorolaMobilityLLC/readme

and put all including the kernel-msm in one folder
then create another motorola folder and inside create "kernel" as simlink to motorola-kernel
and of course inside your kernel-msm/fs folder create f2fs as simlink to ../../motorola/kernel/fs/f2fs/

I can't recall if I did all this myself or got the whole package from somewhere...

lightydo, 2016-02-17:
BTW @mayurdjain you'd better simply clone this repo (kernel-msm) and checkout the branch you need...
apparently the source code tar.gz files @tags07 pointed to, don't have this simlink in them....or just create it yourself and then point directly to motorola-kernel folder.

mayurdjain, 2016-02-18:
Hi Lightydo, 
I am able to build the marshmellow kernel "kernel-msm-MMI-MPI24.65-24" successfully. 
Below are the steps to build the kernel 

1. Download kernel source code from 
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-MPI24.65-24

2. Download Marshmellow FS source code from below link as zip file. 
https://github.com/MotorolaMobilityLLC/motorola-kernel/tree/marshmallow-6.0.0-release

3. Create a folder structure as below 

|-- kernel-msm-MMI-MPI24.65-24    // kernel source 
|-- motorola                                       // fs source marshmallow-6.0.0-release
	|-- Documentation
	|   |-- ABI
	|   `-- filesystems	
	|-- fs
	|   |-- f2fs
	|   `-- Kconfig
	|-- include
	|   |-- linux
	|   `-- trace
	`-- kernel
	    `-- kernel -> /home/../kernel-msm-MMI-MPI24.65-24     //Symbolic link

4. Create a symbolic link f2fs under ../kernel-msm-MMI-MPI24.65-24/fs
/kernel-msm-MMI-MPI24.65-24/fs$  ls -l | grep f2fs
lrwxrwxrwx 1 xxx xxx       xx Feb 18 18:19 f2fs -> /home/../motorola/fs/f2fs

5. update symbolic link  f2fs_fs.h under /kernel-msm-MMI-MPI24.65-24/include/linux

6. update symbolic link f2fs.h under../kernel-msm-MMI-MPI24.65-24/include/trace/events

7.Download tool chain arm-eabi-4.7 
git clone https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/arm/arm-eabi-4.7

8. Build kernel uing below steps 

export CONFIG_FILE="msm8916_defconfig"
export ARCH="arm"
export CROSS_COMPILE="arm-eabi-"
export TOOL_CHAIN_PATH="${HOME}/sources/toolchain/arm-eabi-4.7/bin"
export CONFIG_ABS_PATH="arch/${ARCH}/configs/${CONFIG_FILE}"
export PATH=$PATH:${TOOL_CHAIN_PATH}
make -j16 $CONFIG_FILE
make -j16

9. You can use user msm8916-perf_defconfig if you face any issue with wlan 
export CONFIG_FILE="msm8916-perf_defconfig"

10. I am able to build the kernel successfully with above steps without any issue for marshmellow. 

Thank You!
-Mayur

lightydo, 2016-02-21:
@mayurdjain you're welcome :-)

mayurdjain, 2016-02-21:
@lightydo Thank you for your help and technical support.

casensiom, 2016-06-08:
How to configure the [MMI-MPI24.65-33.1](https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-MPI24.65-33.1) kernel ??

I'm having a lot of compilation errors choosing different defconfig files. Which is the defconfig I need?

> ro.build.id=MPI24.65-33
> ro.build.display.id=MPI24.65-33
> ro.build.version.incremental=34
> ro.build.version.sdk=23
> ro.build.version.preview_sdk=0
> ro.build.version.codename=REL
> ro.build.version.all_codenames=REL
> ro.build.version.release=6.0
> ro.build.version.security_patch=2016-01-01
> ro.build.version.base_os=,release-keys
> ro.build.date=Mon Jan  4 08:03:52 CST 2016
> ro.build.date.utc=1451916232
> ro.build.type=user
> ro.build.user=hudsoncm
> ro.build.host=ilclbld85
> ro.build.tags=release-keys
> ro.build.flavor=osprey_retes-user
> ro.product.model=MotoG3
> ro.product.brand=motorola
> ro.product.name=osprey_retes
> ro.product.device=osprey_umts
> ro.product.board=msm8916

I assume that I would need something like *msm8916_defconfig* or *msm8916-perf_defconfig* but no files with these numbers are found at **arch/arm/configs**

I'm ussing arm-eabi-4.8

Any help from @tags07 or @lightydo will be amazing ;)

lightydo, 2016-06-14:
Hi @casensiom, sorry but only @tags07 can help with the right defconfig...
what are the compilation errors? Can you put an output on gist? (https://gist.github.com/)

charleseb, 2016-06-14:
Hi Carlos and Daniel,

I'm working to push the how to build readme file to github. It will be
available in a few days.

Thanks,
Charles

charleseb, 2017-09-29:
Please refer to readme file here:
https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-MPI24.65-39.txt

closed 2017-09-29 by charleseb