MotorolaMobilityLLC/kernel-msm #55: GPL Source Request for Moto X Force Chinese Version

DevriesL, 2015-12-28:
Hi, I bought a Moto X Force Chinese version also called MOTO X 极 (Ji)
I want to make a custom kernel on these phone
Thank you in advanced

tags07, 2015-12-28:
Could you post the Software Version/Build Version you have on the device under Settings->About Phone. I'll check if that's already posted or if its in our pipieline

DevriesL, 2015-12-29:
Hi Palaniappan,
System version: 23.1.229.kinzie_retcn_ds.retcn.en.CN retcn
Build number: LPK23
Kernel version: 3.10.49-perf-g02cfaca

tags07, 2015-12-29:
LPK23.145-5 is in our pipeline. We'll get to posting it soon.

DevriesL, 2016-01-02:
Hi Palaniappan,
Sorry I have made a mistake
The System version: 23.11.1.kinzie_retcn_ds.retcn.en.CN retcn
Build number: LPK23.229
Kernel version: 3.10.49-perf-g02cfaca

tags07, 2016-01-06:
@DevriesL I don't see any record of us shipping LPK23.229
I'll check internally and post the sources if we are shipping it.

DevriesL, 2016-01-07:
OK, Thank you, hope to get this kernel source :D

DevriesL, 2016-02-07:
@tags07 It is been really a long time...

charleseb, 2016-02-07:
Hi DevriesL,

I'm working to release this tag.
This version will be available in a few days.

Thanks,
Charles

charleseb, 2016-02-15:
Hi all,

The version MMI-LPK23.229 has been released to Github.

Thanks,
Charles

closed 2016-02-15 by charleseb