MotorolaMobilityLLC/kernel-msm #57: Enable notification LED on Moto X Play

amandal1810, 2016-01-02:
Kindly enable the notification LED which already exists on the Moto X Play (lux) but has been kept disabled.

luk1337, 2016-01-02:
I don't think this is the right place. Also they keep it disabled for reason I guess.

piratenpanda, 2016-01-28:
Didn't notice any problems by using it so far

piratenpanda, 2016-04-12:
Is the any reason why the new sensor hub firmware disables the led?

awheeler-mmi, 2017-10-02:
Hey all,

On Lynx/lux we replaced general LED notification support with Active Display. The sensorhub firmware that disabled LED support enabled our glance (face/attention detection) sensor.  After that update the LED is just used to indicate that the battery is extremely low, etc.  We can't roll out a new firmware that re-enables LED, but if reverting the firmware works then great.

closed 2017-10-02 by awheeler-mmi