MotorolaMobilityLLC/kernel-msm #58: I'm having problems with wifi driver lollipop-5.1.1-release-ghost

Klozz, 2016-01-12:
D/WifiService(  946): setWifiEnabled: true pid=4178, uid=1000
E/WifiService(  946): Invoking mWifiStateMachine.setWifiEnabled
E/WifiStateMachine(  946): setting operational mode to 1
E/WifiHW  (  946):  User build,dont Start logging service.
E/WifiStateMachine(  946): Failed to load driver
E/WifiStateMachine(  946): setWifiState: unknown state
I/MotoNetwCtrlr( 1631): onReceive: Completed intent=Intent { act=android.net.wifi.WIFI_STATE_CHANGED flg=0x4000010 (has extras) } Action=android.net.wifi.WIFI_STATE_CHANGED
E/WifiStateMachine(  946): setSuspendOptimizations: 2 true
E/WifiStateMachine(  946): mSuspendOptNeedsDisabled 0

I forked from https://github.com/MotorolaMobilityLLC/vendor-qcom-opensource-wlan-prima
Any help will be appreciated :)

My tag is
XT1058-
LPA23.12-21.7

tags07, 2016-02-10:
How did you build the driver?
did you build with something like this:
`make -C kernel KBUILD_RELSRC=$kernel_dir M=../vendor/qcom/opensource/wlan/prima O=$KERNEL_OBJ ARCH=arm CROSS_COMPILE=arm-eabi- modules WLAN_ROOT=../vendor/qcom/opensource/wlan/prima MODNAME=wlan BOARD_PLATFORM=msm8960 CONFIG_PRIMA_WLAN=m`

Klozz, 2016-02-10:
yes this form and this other:
Adding wlan prima inside drivers/Staging and added in defconfig CONFIG_PRIMA_WLAN=m inside msm8960 defconfig
The driver is generated, but for some reason he can not read.
Thanks for answering

charleseb, 2017-09-29:
Closing old issue.
Kernel command to wlan above.

closed 2017-09-29 by charleseb