MotorolaMobilityLLC/kernel-msm #60: hab check failed for custom compiled zImage - XT1097

glauberlima, 2016-01-16:
Hi everyone.

I've an unlocked Moto X (2014) XT1097.

Despite being able to boot via "fastboot boot myCompiledKernel.img" I get this error message when I flash the boot image (fastboot flash boot myCompiledKernel.img:
>> hab check failed for boot <<

And the phone won't boot - it's fallbacks to bootloader.

I've been using abootimg to put my compiled zImage inside the original untouched Moto boot.img.

I'm wondering what step I've been doing wrong....

Any help is appreciated.

TalAloni, 2016-06-23:
A successful flash of a self compiled (read: invalid signature) kernel will always print "hab check failed for boot". (Tested with my XT1032).
Sounds like you flashed successfully and have an issue with your kernel.

awheeler-mmi, 2017-09-27:
This one is pretty old, so I'll close it.

closed 2017-09-27 by awheeler-mmi