MotorolaMobilityLLC/kernel-msm #62: Titan: MPB24.65-34

luk1337, 2016-01-23:
System Version: 24.41.34.titan_retaildsds.retaildsdsall.en.03
Build Number: MPB24.65-34
Kernel version: Linux version 3.4.42-gef77327 (hudsoncm@ilclbld53) (gcc version 4.8 (GCC) ) #1 SMP PREEMPT Wed Jan 6 12:29:54 CST 2016

luca020400, 2016-02-09:
Bump
This build has been officially released in India

luk1337, 2016-02-10:
@tags07 @charleseb ;_;

tags07, 2016-02-10:
@luk1337 : Yes, looks like we got approval to ship this mid Jan 2016. I need to check if we completed the rollout to all regions. We have this in our pipeline. We'll get to it soon.

luca020400, 2016-02-10:
Only India as of now ....

luk1337, 2016-02-10:
yeah, XT1068 and XT1079 got it already...

luca020400, 2016-02-10:
@tags07 as far as I know thea received same build as titan ( MPB24.65-34 )
When MPB24.65-20 has been released ?
Thanks

charleseb, 2016-02-10:
Hi Luca,

I just released MMI-MPB24.65-20 to github.

Thanks,
Charles

luca020400, 2016-02-10:
What about -34 ? :p

tags07, 2016-02-10:
@luca020400 we'll post that soon.. we have a huge backlog and Charles is multi tasking with other priorities. I'll sync up with him and prioritize posting all commits till the latest tags we shipped.

luca020400, 2016-02-10:
Great thank you both :)

archon810, 2016-03-01:
@charleseb @tags07 Can you confirm that this update has started rolling out in US now please? We got a report from a user who isn't in the soak test and is using a US model.

Thanks.

luca020400, 2016-03-01:
@archn810: as far as I know XT1064 ( us ) didn't receive it yet
Only XT1068 ( India )

archon810, 2016-03-01:
@luca020400 Got 2 reports already in the last few hours of the US version.

luca020400, 2016-03-01:
Oh nice, we're still waiting in EU ..

luca020400, 2016-03-01:
@tags07 @charleseb Since this issue has been bumped . Any news on this release ? Thanks

tags07, 2016-03-01:
@luca020400  I am slacking behind a bit on the list of releases to post. I'll check and add it to the pipeline.

charleseb, 2016-03-03:
Hi @luca020400,

I'm working to release this version. It will be available in a few days.

Thanks,
Charles

luca020400, 2016-03-03:
@charleseb thanks :)
@archon810 Looks like EU received some M love too ;) ( XT1072 - moto g 2014 lte )

luca020400, 2016-03-07:
@charleseb thanks

closed 2016-03-07 by luk1337

charleseb, 2016-03-07:
Hi  @luca020400,

The version MMI-MPB24.65-34 has been released to Github.

Thanks,
Charles

luca020400, 2016-03-07:
@charleseb Just for curiosity, did you ever ship -20 ?
I couldn't find a device using it

tags07, 2016-03-07:
@luca020400 How do you know we didnt ship?

luca020400, 2016-03-07:
@tags07 I checked every titan and thea 6.0 release :P

tags07, 2016-03-07:
@luca020400 Do you work for Lenovo/Motorola? If not, I'd like access to that database you are checking..sometimes its a pain internally to see what we shipped ;-)

luk1337, 2016-03-07:
@tags07
No, he doesn't work for Moto/Lenovo but there's like filefactory folder of guy posting like all factory images for the whole G series xD

luca020400, 2016-03-08:
@tags07: http://www.filefactory.com/folder/c6cdedc45a775d27/?sort=created&order=DESC&show=100&page=1
Enjoy downloading at 40 kb/s ...