MotorolaMobilityLLC/kernel-msm #64: Source request for Merlin (Moto G Turbo Edition)

omerjerk, 2016-02-12:
Hi,
I plan to develop on Moto G Turbo Edition.
By when can we expect the kernel source for this device to be available ?

luca020400, 2016-02-12:
I think the kernel tree for merlin is MMI-MPI24.65-24 tag ( msm8916 and msm8939 shares the same kernel tree )
@tags07

omerjerk, 2016-02-12:
So which exactly which defconfig should I use ?

luca020400, 2016-02-12:
I couldn't find a proper config for marlin
Probably they have to release it yet

tags07, 2016-02-12:
MPI24.65-24:  msm8916-perf_defconfig

If you tell me what tag you are looking for, I can tell you the defconfig
or if the tag is still in our pipeline..

omerjerk, 2016-02-12:
@tags07 
So just reconfirming, I should use `msm8916-perf_defconfig` at the tag `MPI24.65-24` for building the kernel for Moto G Turbo Edition ?

luca020400, 2016-02-12:
If you are running MPI24.65-24 build yes
Otherwise post your build version

tags07, 2016-02-12:
MPI24.65-24 is for osprey and you can use msm8916-perf_defconfig  for
defconfig. If you are looking for merlin, I am checking..we may have posted
Lollipop release but Android M may be in pipeline..

omerjerk, 2016-02-12:
I think this should do. :)
Thanks for the help.
P.S. My build version is MPD24.65-22

closed 2016-02-12 by omerjerk