MotorolaMobilityLLC/kernel-msm #70: Marshmallow Kernel source for titan

cmd1084, 2016-04-10:
I can't find a defconfig in the Kernel source. Please be Patient. I am new to this.

luca020400, 2016-04-10:
@tags07 You might add a readme with the guide to build the kernel for titan aswell
@cmd1084 the main defconfig is arch/arm/configs/msm8226_defconfig and the extconfig is arch/arm/configs/ext_config/titan-defconfig.config. Combine them with
cat arch/arm/configs/ext_config/titan-defconfig.config >> arch/arm/configs/msm8226_defconfig
make msm8226_defconfig
make

cmd1084, 2016-04-10:
Thank you luca

tags07, 2016-04-10:
We are working on generating a standalone build instruction with each posting. Stay tuned. That should simplify things.

luca020400, 2016-04-10:
@tags07 nice idea, it will help a lot of users

closed 2017-09-22 by charleseb