MotorolaMobilityLLC/kernel-msm #72: Kernel source request for Moto MAXX (quark) MM

18712886438, 2016-05-21:
Looking forward to the MM kernel source of quark. Thx. :+1:

tags07, 2016-05-27:
Could you post the System Version and Build Number of the sources you are looking for?

18712886438, 2016-05-28:
@tags07 MPG24.107-70.1  android 6.0

tags07, 2016-05-28:
Currently all commits leading upto MPG24.107-70.2 is in our pipeline. We'll
start rolling them out in the coming weeks.

18712886438, 2016-05-28:
@tags07  thanks :)

18712886438, 2016-06-08:
@tags07    any update？

charleseb, 2016-06-08:
Hi CallMESuper,

I'm still working on this tag. It will be released soon.

Thanks,
Charles

18712886438, 2016-06-08:
@charleseb  Thx  :+1:

charleseb, 2016-06-09:
Hi CallMESuper,

The version MMI-MPG24.107-70.2 is released to Github.

https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-MPG24.107-70.2

Thanks,
Charles

closed 2016-06-09 by charleseb