MotorolaMobilityLLC/kernel-msm #73: Kernel source code for Moto G4 Plus

yeshwanthvshenoy, 2016-05-23:
Please release the kernel source code for Motorola G4 Plus(Athene).

tags07, 2016-05-27:
Please post the System Version/Build Number of the device SW and we'll post the source soon.

yeshwanthvshenoy, 2016-05-28:
@tags07 
Build Number : MPJ24.139-23.1
Android Version : 6.0.1
Codename : Athene

tags07, 2016-05-28:
Thankyou, currently posting all commits leading up to MPJ24.139-23.4 is in the works. Stay tuned.

yeshwanthvshenoy, 2016-05-28:
@tags07 Thanks :) We are waiting

yeshwanthvshenoy, 2016-05-31:
@tags07 Any update on this item?

charleseb, 2016-05-31:
Hi Yeshwanth,

I'm working to release this version and it will be available in a few days.

Thanks,
Charles

yeshwanthvshenoy, 2016-06-01:
@charleseb thanks :+1:

pranav01, 2016-06-07:
@tags07  Any update with release?

charleseb, 2016-06-07:
Hi Pranav,

I'm still working on this version. It will be released soon.

Thanks,
Charles

yeshwanthvshenoy, 2016-06-08:
Thanks. Closing the issue now.

closed 2016-06-08 by yeshwanthvshenoy

yeshwanthvshenoy, 2016-06-15:
@charleseb Could you please point out which defconfig is used for compiling the stock kernel? Is it msm8916_defconfig or msm_defconfig? There seems to be commits related to msm8952 under msm_defconfig.

charleseb, 2016-06-15:
Hi Yeshwanth,

I'm working to release the readme file with the compile process related to
this tag. It will be released soon.

Thanks,
Charles

yeshwanthvshenoy, 2016-06-16:
@charleseb Thanks! I tried to compile my own kernel but it ended up in bootloops and stuck on moto boot logo. Could be great if you can provide a README for this! Thanks!

ramsudharsan, 2016-06-19:
@charleseb @tags07 https://github.com/ramsudharsan/android_kernel_motorola_msm8952/commit/91242f4c18799ab73d5031fcd75d06bf1f6e8ef7

C-Aniruddh, 2016-06-27:
@yeshwanthvshenoy you could pull the kernel config from a running device.

`adb pull /proc/config.gz`

HelllGuest, 2016-06-27:
@yeshwanthvshenoy  unpack stock boot image and use extract-ikconfig (from kernel_dir/scripts) to get stock defconfig. eg: ./extract-ikconfig zImage > stock_defconfig . rename stock_defconfig .config and compile.