MotorolaMobilityLLC/kernel-msm #75: Kernel source for Moto G Turbo (moto g3-te) [merlin]

psndna88, 2016-05-25:
requesting release of kernel  source for 6.0.1 Moto G3-TE merlin xt-1556/7 ...

tags07, 2016-05-27:
Please post the Build Number/System Version of the device. I'll check if we have completed the shipment   with the requested version and post the sources.

psndna88, 2016-05-27:
BUILD. PROP:
# begin build properties # autogenerated by buildinfo.sh ro.build.id=MPD24.107-56 ro.build.display.id=MPD24.107-56 ro.build.version.incremental=30 ro.build.version.sdk=23 ro.build.version.preview_sdk=0 ro.build.version.codename=REL ro.build.version.all_codenames=REL ro.build.version.release=6.0.1 ro.build.version.security_patch=2016-03-01 ro.build.version.base_os= ro.build.date=Thu Feb 25 17:47:23 CST 2016 ro.build.date.utc=1456444043 ro.build.type=user ro.build.user=hudsoncm ro.build.host=ilclbld35 ro.build.tags=release-keys ro.build.flavor=merlin_retasia-user ro.product.model=MotoG3-TE ro.product.brand=motorola ro.product.name=merlin_retasia ro.product.device=merlin ro.product.board=msm8916 # ro.product.cpu.abi and ro.product.cpu.abi2 are obsolete, # use ro.product.cpu.abilist instead. ro.product.cpu.abi=armeabi-v7a ro.product.cpu.abi2=armeabi ro.product.cpu.abilist=armeabi-v7a,armeabi ro.product.cpu.abilist32=armeabi-v7a,armeabi ro.product.cpu.abilist64= ro.product.manufacturer=motorola ro.product.locale=en-GB ro.wifi.channels= ro.board.platform=msm8916 # ro.build.product is obsolete; use ro.product.device ro.build.product=merlin # Do not try to parse description, fingerprint, or thumbprint ro.build.description=merlin_retasia-user 6.0.1 MPD24.107-56 30 release-keys ro.build.fingerprint=motorola/merlin_retasia/merlin:6.0.1/MPD24.107-56/30:user/release-keys ro.build.characteristics=default # end build properties # # from device/qcom/msm8916_32/system.prop # # # system.prop for msm8916 # #rild.libpath=/system/lib/libreference-ril.so rild.libpath=/system/vendor/lib/libril-qc-qmi-1.so rild.libargs=-d /dev/smd0 persist.rild.nitz_plmn= persist.rild.nitz_long_ons_0= persist.rild.nitz_long_ons_1= persist.rild.nitz_long_ons_2= persist.rild.nitz_long_ons_3= persist.rild.nitz_short_ons_0= persist.rild.nitz_short_ons_1= persist.rild.nitz_short_ons_2= persist.rild.nitz_short_ons_3= persist.radio.rat_on=combine ril.subscription.types=NV,RUIM DEVICE_PROVISIONED=1 # Start in cdma mode # Motorola set later #ro.telephony.default_network=5 debug.sf.hw=1 debug.egl.hw=1 persist.hwc.mdpcomp.enable=true debug.mdpcomp.logs=0 dev.pm.dyn_samplingrate=1 persist.demo.hdmirotationlock=false debug.enable.sglscale=1 #ro.hdmi.enable=true #tunnel.decode=true #tunnel.audiovideo.decode=true #lpa.decode=false #lpa.use-stagefright=true #persist.speaker.prot.enable=false # # system props for the cne module # persist.cne.feature=0 # # system props for the iwlan module # persist.data.iwlan.enable=false #system props for the MM modules media.stagefright.enable-player=true media.stagefright.enable-http=true media.stagefright.enable-aac=true media.stagefright.enable-qcp=true media.stagefright.enable-fma2dp=true media.stagefright.enable-scan=true media.msm8939hw=0 media.msm8929hw=0 mmp.enable.3g2=true media.aac_51_output_enabled=true #codecs: DivX DivXHD AVI AC3 ASF AAC QCP DTS 3G2 MP2TS FLAC mm.enable.qcom_parser=3314291 # Default to AwesomePlayer media.stagefright.use-awesome=false # # system props for the data modules # ro.use_data_netmgrd=true persist.data.netmgrd.qos.enable=true #system props for time-services persist.timed.enable=true # System property for cabl ro.qualcomm.cabl=0 # # System props for telephony # System prop to turn on CdmaLTEPhone always # Motorola default to 0 telephony.lteOnCdmaDevice=0 # # System props for bluetooh # System prop to turn on hfp client bluetooth.hfp.client=1 #Simulate sdcard on /data/media # persist.fuse_sdcard=true # #snapdragon value add features # ro.qc.sdk.audio.ssr=false ##fluencetype can be "fluence" or "fluencepro" or "none" ro.qc.sdk.audio.fluencetype=fluence persist.audio.fluence.voicecall=true persist.audio.fluence.voicerec=false persist.audio.fluence.audiorec=false persist.audio.fluence.speaker=false persist.audio.fluence.mode=endfire ro.config.vc_call_vol_steps=8 # Tone generator maximum volume be between 0 and 100 (0dB) ro.audio.ringbeep_volume=73 #Buffer size in kbytes for compress offload playback audio.offload.buffer.size.kb=240 #Minimum duration for offload playback in secs audio.offload.min.duration.secs=30 # Motorola enable offload audio playback audio.offload.disable=0 #Enable offload audio video playback by default av.offload.enable=true #enable voice path for PCM VoIP by default use.voice.path.for.pcm.voip=false # #System property for FM transmitter # ro.fm.transmitter=false #enable dsp gapless mode by default audio.offload.gapless.enabled=true #Audio voice concurrency related flags voice.playback.conc.disabled=true voice.record.conc.disabled=true voice.voip.conc.disabled=true #property to enable user to access Google WFD settings persist.debug.wfd.enable=0 #propery to enable VDS WFD solution persist.hwc.enable_vds=0 #selects CoreSight configuration to enable persist.debug.coresight.config=stm-events #property to enable narrow search range for video encoding vidc.enc.narrow.searchrange=1 #property to enable DS2 dap audio.dolby.ds2.enabled=true #Trim properties ro.sys.fw.use_trim_settings=true ro.sys.fw.empty_app_percent=50 ro.sys.fw.trim_empty_percent=100 ro.sys.fw.trim_cache_percent=100 ro.sys.fw.trim_enable_memory=1073741824 #Factory Reset Protection #ro.frp.pst=/dev/block/bootdevice/by-name/config #min/max cpu in core control ro.core_ctl_min_cpu=2 ro.core_ctl_max_cpu=4 # # ADDITIONAL_BUILD_PROPERTIES # ro.build.version.full=Blur_Version.24.61.56.merlin_retasia.retasia.en.03 ro.mot.build.customerid=retasia ro.mot.build.version.sdk_int=24 ro.product.display=Moto  [... cut ...] st.rcs.supported=0 persist.rcs.presence.provision=0 persist.ims.disableADBLogs=2 persist.ims.disableDebugLogs=0 persist.ims.disableQXDMLogs=0 persist.ims.disableIMSLogs=0 persist.sys.dalvik.vm.lib.2=libart dalvik.vm.isa.arm.variant=cortex-a53 dalvik.vm.isa.arm.features=default net.bt.name=Android dalvik.vm.stack-trace-file=/data/anr/traces.txt persist.gps.qc_nlp_in_use=0 ro.gps.agps_provider=1 ro.pip.gated=0 ro.expect.recovery_id=0xa70808ef30b687e1c640feb233bd8bf211238ae4000000000000000000000000

tags07, 2016-05-28:
Thank you. We are working on posting MPD24.107-56. Stay tuned.

charleseb, 2016-06-08:
HI @psndna88
Tag released.
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-MPD24.107-56

Thanks,
Charles

closed 2016-06-08 by charleseb