MotorolaMobilityLLC/kernel-msm #79: ... is not clean. Please run make mrproper

SpasilliumNexus, 2016-06-25:
I was following directions on how to compile the kernel for clark from here: https://raw.githubusercontent.com/MotorolaMobilityLLC/readme/master/MMI-LPH23.116-18.1.txt

Everything seemed to work fine with building, but now when I try building again, I'm getting an error and to run make mrproper: http://pastebin.com/kSy3r896

I've tried make mrproper inside the /aosp-motorola/kernel directory, but it comes back with 

> Makefile:803: *** multiple target patterns.  Stop.

I've also tried make mrproper inside the /aosp-motorola directory, but that comes back with 

> make: *** No rule to make target 'mrproper'.  Stop.

Could anyone tell me what command I need to use in order to clean and rebuild?

Thank you.

tags07, 2016-06-25:
you can do : git clean -xfd   
that would clean the git.

SpasilliumNexus, 2016-06-25:
Thank you! That works perfectly.

closed 2016-06-25 by SpasilliumNexus