MotorolaMobilityLLC/kernel-msm #80: Source built zImage (or) DT fails to boot (Athene)

ashwinr64, 2016-06-26:
Using prebuilt DT and Zimage built out of 'marshmallow-6.0.1-release-athene' branch , it fails to boot the stock ROM which has the release number 'MPJ24.139-13.1' . It was built with GCC 4.9 . 
Also,
While using a combination of prebuilt ZImage and Source built DT , phone stays at Moto Logo.
Source to dtbtool: https://github.com/CyanogenMod/android_device_qcom_common/tree/cm-13.0/dtbtool
I tried diffing the prebuilt DT image and my source built dt image ,They both are extremely different.I wonder if you were using a dtbtool slightly deviated from CodeAurora .If so, kindly share the code.

P.S: I'm not able to log anything.Kindly help.

luca020400, 2016-06-26:
cat arch/arm/configs/ext_config/moto-msm8916.config >> arch/arm/configs/msm8916-perf_defconfig    
make dtbs
I just created dt.img locally and it's exactly the same as the one shipped in moto boot.img

yeshwanthvshenoy, 2016-06-28:
@luca020400 I tried what you mentioned but I ended up with something else when doing the make msm8916-perf_defconfig itself. Here is what happened : http://pastebin.com/8R4X33R0

luca020400, 2016-06-28:
You need to clone motorola-kernel in $kernel_tree_path/../.

yeshwanthvshenoy, 2016-06-28:
Ahhh okay, I guess we missed that. Cool! Thanks! :)

ashwinr64, 2016-06-28:
F2fs code in motorola-kernel has been imported to the kernel, so motorola-kernel is not needed in my case.

luca020400, 2016-07-29:
@tags07 @charleseb can you release the dtbTool used in official athena builds? 
@ashwinr64 athena isn't using a standard dt.img format, there is a extra field ( hence the the /wrong/ 259 version )

ashwinr64, 2016-07-30:
That's what I was precisely saying. Kindly update us, Motorola developers ....

charleseb, 2016-08-09:
Hi Ashwin, Luca and all

You can reach the prebuilt dtbTool related to version MMI-MPJ24.139-23.4 at
https://github.com/MotorolaMobilityLLC/readme/tree/master/dtbtool_MMI-MPJ24.139-23.4

Thanks,
Charles

luca020400, 2016-08-09:
@charleseb any plan to release the source code? The difference between caf and moto is relatively small, I don't think there is any reason to share it as prebuilt

ashwinr64, 2016-08-10:
Thanks for the hospitality @charleseb !

ashwinr64, 2016-08-19:
Is this still on @charlesb ?

charleseb, 2016-08-19:
Hi Ashwin,

I'm still working to release this code. I'll update in a few days.

Thanks,
Charles

ashwinr64, 2016-08-19:
That's nice to hear @charlesb . Waiting for the same.

Vachounet, 2017-01-18:
Does the new N release needs an update dtbtool ?

Builded a fresh boot : 
    -If i try to flash it with fastboot, error : Too many links
    -If i try to boot it with fastboot, stuck at Moto logo
    -If a try to use dd, it just soft brick my device.

Something i missed here ?

rahulsnair, 2017-01-18:
i have checked too. Gets the same error.

charleseb, 2017-01-18:
Hi all,

You can use this prebuilt dtbTool at https://github.com/MotorolaMobilityLLC/readme/tree/master/dtbtool_MMI-NPJ25.93-11

Thanks,
Charles

Vachounet, 2017-01-19:
Thanks @charleseb

closed 2017-01-24 by charleseb