MotorolaMobilityLLC/kernel-msm #81: G4 Plus "Athene" WiFi not working with stock ROM

flar2, 2016-07-22:
I can build and boot the kernel fine, using config pulled from stock kernel.
I can also build and load the wlan module using vendor-qcom-opensource-wlan-prima

But WiFi will not work.  

I thought the problem was firmware, so I've also tried loading the stock wlan module with my kernel (with CONFIG_MODULE_FORCE_LOAD off so it loads without signature).  Same result.  Module can be loaded but WiFi will not work.

Is there any way WiFi can work with the stock ROM?   Preferably, I would just like to load the stock wlan module with my kernel.

attack11, 2016-07-22:
Are you using GCC 4.9 ?

[https://github.com/Alberto97/android_kernel_motorola_msm8916/commit/ab12d38e17ef5371e20fe92c361fda6ff514a73f](url)

flar2, 2016-07-22:
@attack11 
Thanks, but the module loads just fine.  WiFi just gets stuck trying to turn on.

BTW, I'm using the standard GCC 4.8 prebuilt toolchain

flar2, 2016-07-22:
I've now tried building the kernel with GCC 4.9, since that is what the stock kernel is built with, again using the config pulled from stock kernel.

It will not even boot.

@motorolamobility 
Could we get some instructions or more information?

flar2, 2016-07-23:
I ended up going back to GCC 4.8, signing the stock wlan module for my kernel and mounting an image at /system/lib/modules.  WiFi worked after this, but it's not a very good solution since I will likely break the wlan module if I alter the kernel too much.

This is not a very good solution, so any additional info would be appreciated.

HelllGuest, 2016-07-23:
What about  CONFIG_MODULE_WHITELIST=n ?

18712886438, 2016-07-24:
I also met this problem on my device. When I insmod wlan.ko, there will  be an error about load modules according  to kmsg due to modules hashcode mismatich, Just turn off CONFIG_MODULE_WHITELIST. :)

flar2, 2016-07-25:
I never had any trouble inserting the module.  The module loads but wifi will not start.  My issue is like this one:  https://github.com/MotorolaMobilityLLC/readme/issues/10

But anyway, I'm happy with my current solution.

HelllGuest, 2016-07-26:
For the **_CONFIG_MODULE_WHITELIST_** mechanism module hash stored in this file  **_module_hashes_** in ramdisk, used by  **_init.mmi.rc_**  

> copy /module_hashes /sys/kernel/module_whitelist

elgithubo, 2016-08-02:
@flar2 I am currently trying to get wlan work for cm13 on athene.
Might you please shed some words about how you could get the stock wlan.ko to work with your compiled kernel.
I have the error "disagrees about version of symbol module_layout" in dmesg when trying to insert, and even if I change module_layout in hex editor to make the version match, I cannot insmod wlan.ko and get spawned with tons of 'disagrees about version of symbol' errors.
Any help would be really appreciated.

yeshwanthvshenoy, 2016-08-03:
@elgithubo I don't think he made changes to make it work. If the device is rooted, it works. No one has an idea why. Am I right @flar2 ? We need to know how to the dtb is made by Motorola for Athene. This isn't the way how it is usually done.

flar2, 2016-08-03:
@elgithubo @yeshwanthvshenoy 
Just to clarify, I am able to use the stock wlan module because I don't make extensive changes to the rest of the kernel, so module_layout and CRCs for other functions used by the module stay the same. I also re-signed the module to work with my kernel. 

Root is not actually needed for the module to load.  I chose to mount an image with the re-signed stock modules at /system/lib/modules, and this requires root due to selinux restrictions.  I only do this because I don't want to modify users' system partition, to ensure they can still receive OTA updates or return to the stock kernel with ease.

elgithubo, 2016-08-11:
@flar2 
You might want to check @ashwinr64's trees. He managed to compile a working wlan from source.

ejjohnson1999, 2016-12-09:
has there been any new development on this? I'm building for stock rom and am not having any luck getting wifi working

closed 2018-04-08 by flar2