MotorolaMobilityLLC/kernel-msm #82: Droid X/ Verizon griffin source

CunningLogic, 2016-08-10:
Can someone push the relevant branch for this device please

[ro.build.fingerprint]: [motorola/griffin_verizon/griffin:6.0.1/MCL24.203-22/24:user/release-keys]

Linux localhost 3.18.24-perf-g4f3d148-00001-g2c82a6a #1 SMP PREEMPT Fri Jun 10 20:55:29 CDT 2016 aarch64

luk1337, 2016-08-10:
You likely want to post build description / fingerprint too. Otherwise someone will reply to this after 2 weeks asking you to do so...

CunningLogic, 2016-08-10:
thanks

[ro.bootimage.build.date]: [Fri Jun 10 20:53:17 CDT 2016]
[ro.bootimage.build.date.utc]: [1465609997]
[ro.bootimage.build.fingerprint]: [motorola/griffin_verizon/griffin:6.0.1/MCL24.203-22/24:user/release-keys]
[ro.build.characteristics]: [default]
[ro.build.date]: [Fri Jun 10 20:48:39 CDT 2016]
[ro.build.date.utc]: [1465609719]
[ro.build.description]: [griffin_verizon-user 6.0.1 MCL24.203-22 24 release-keys]
[ro.build.display.id]: [MCL24.203-22]
[ro.build.fingerprint]: [motorola/griffin_verizon/griffin:6.0.1/MCL24.203-22/24:user/release-keys]
[ro.build.flavor]: [griffin_verizon-user]
[ro.build.host]: [ilclbld32]
[ro.build.id]: [MCL24.203-22]
[ro.build.product]: [griffin]
[ro.build.tags]: [release-keys]
[ro.build.thumbprint]: [6.0.1/MCL24.203-22/24:user/release-keys]
[ro.build.type]: [user]
[ro.build.user]: [hudsoncm]
[ro.build.version.all_codenames]: [REL]
[ro.build.version.base_os]: []
[ro.build.version.codename]: [REL]
[ro.build.version.full]: [Blur_Version.24.14.24.griffin_verizon.verizon.en.US]
[ro.build.version.incremental]: [24]
[ro.build.version.preview_sdk]: [0]
[ro.build.version.qcom]: [LA.HB.1.3.2-12000-8x96.0]
[ro.build.version.release]: [6.0.1]
[ro.build.version.sdk]: [23]
[ro.build.version.security_patch]: [2016-05-01]
[ro.mot.build.customerid]: [verizon]
[ro.mot.build.device]: [griffin]
[ro.mot.build.oem.product]: [griffin_verizon]
[ro.mot.build.product.increment]: [14]
[ro.mot.build.system.product]: [griffin_verizon]
[ro.mot.build.version.release]: [24.14]
[ro.mot.build.version.sdk_int]: [24]

CunningLogic, 2016-08-10:
[ro.bootimage.build.date]: [Fri Jun 10 20:53:17 CDT 2016]
[ro.bootimage.build.date.utc]: [1465609997]
[ro.bootimage.build.fingerprint]: [motorola/griffin_verizon/griffin:6.0.1/MCL24.203-22/24:user/release-keys]
[ro.build.characteristics]: [default]
[ro.build.date]: [Fri Jun 10 20:48:39 CDT 2016]
[ro.build.date.utc]: [1465609719]
[ro.build.description]: [griffin_verizon-user 6.0.1 MCL24.203-22 24 release-keys]
[ro.build.display.id]: [MCL24.203-22]
[ro.build.fingerprint]: [motorola/griffin_verizon/griffin:6.0.1/MCL24.203-22/24:user/release-keys]
[ro.build.flavor]: [griffin_verizon-user]
[ro.build.host]: [ilclbld32]
[ro.build.id]: [MCL24.203-22]
[ro.build.product]: [griffin]
[ro.build.tags]: [release-keys]
[ro.build.thumbprint]: [6.0.1/MCL24.203-22/24:user/release-keys]
[ro.build.type]: [user]
[ro.build.user]: [hudsoncm]
[ro.build.version.all_codenames]: [REL]
[ro.build.version.base_os]: []
[ro.build.version.codename]: [REL]
[ro.build.version.full]: [Blur_Version.24.14.24.griffin_verizon.verizon.en.US]
[ro.build.version.incremental]: [24]
[ro.build.version.preview_sdk]: [0]
[ro.build.version.qcom]: [LA.HB.1.3.2-12000-8x96.0]
[ro.build.version.release]: [6.0.1]
[ro.build.version.sdk]: [23]
[ro.build.version.security_patch]: [2016-05-01]
[ro.mot.build.customerid]: [verizon]
[ro.mot.build.device]: [griffin]
[ro.mot.build.oem.product]: [griffin_verizon]
[ro.mot.build.product.increment]: [14]
[ro.mot.build.system.product]: [griffin_verizon]
[ro.mot.build.version.release]: [24.14]
[ro.mot.build.version.sdk_int]: [24]

charleseb, 2017-01-19:
Hi all,

Version MMI-MCL24.203-22 released.
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-MCL24.203-22

closed 2017-01-19 by charleseb