MotorolaMobilityLLC/kernel-msm #85: Quark 4g VoLTE support

z2z, 2016-10-03:
Support for 4g VoLTE disabled in Marshmallow 6.0.1 ?

> MMI-MPG24.107-70.2
> https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-MPG24.107-70.2

bhb27, 2016-10-03:
That version was released to XT1225 only.
Quark has 3 model and XT1225 doesn't have VoLTE.

Verizon model XT1254 has VoLTE the last kernel released to that was:
https://github.com/MotorolaMobilityLLC/kernel-msm/tree/lollipop-5.1.1-release-quark

Any way I have try to enable VoLTE in XT1225 using that 5.1.1 source VoLTE doesn't seems to be a part of the kernel.

I don't know how to have it fully working, but will be nice to have a Motorola answer about this if the kernel has anything to do with it, unfortunately this type of answer usually don't receive answer by Motorola...

**Motorola should answer all the support request related to kernel/software here in they git, after all we are they consumer asking very specif kernel/software related question others support channel in most situation have no clue how to  answer the questions ask here...**

awheeler-mmi, 2016-10-03:
> release-quark

Hey Felipe,

We don't have a VoLTE-specific kernel.  You should be able to get VoLTE to
work with this kernel, though perhaps there are some init changes in the
ramdisk that are needed.  Maybe look around for an extracted ramdisk from
XT1254?

You're also correct in that we don't officially provide direct support for
these sorts of experiments.  But we do welcome them!!

-andrew

bhb27, 2016-10-04:
Thanks for the answer awheeler-mmi,

I did go trow all init files check the 
init.mmi.volte.rc
/system/bin/rcs_config.sh
+ any other related to this...

It seems to me that kernel, init, libs, framworks and prop flags is not the problem.
Is some thing related to the hardware hall that need to be build change to integrate yours files to the ROM, I get that because when everything Motorola/verizon did that can be see and use I did/use but the  featuring doesn't work.

I keep trying :+1:

z2z, 2016-10-04:
@bhb27  ..yes writing for XT1225
> That version was released to XT1225 only.
Quark has 3 model and XT1225 doesn't have VoLTE.

So essentially your saying XT1225 hardware doesn't support VoLTE? or its kernel/software level? 

Some Ref: ( I think XT1225 has VoLTE hardware level support)
https://www.qualcomm.com/products/snapdragon/processors/805
http://flipshope.com/blog/how-to-check-enable-volte-support-in-any-mobile/

bhb27, 2016-10-04:
@z2z  when I referring to XT1225 not having VoLTE is in the software side, I on a XT1225 were I live no carrier supports it, for a example not even LTE is available were I live from my carrier.

If you wanna to have full VoLTE support the key here is to understand how Motorola/Verizon is making it work, the kernel seems to make no difference.

If you manage to achieve VoLTE in a non Verizon stock ROM or a ROM base on that stock let me know...

z2z, 2016-10-11:
@bhb27  Radio?

@awheeler-mmi / MotorolaMobilityLLC
Can you help? ... As suggested by @bhb27 Customer support know very little about this things.

luca020400, 2016-10-11:
@z2z Did you check if volte it's enabled framework side?
These are the needed configs to enable it android side ( plus ims blobs, but moto should've shipped them )
````xml
<bool name="config_device_volte_available">true</bool>
<bool name="config_carrier_volte_available">true</bool>
<bool name="config_device_vt_available">true</bool>
<bool name="config_carrier_vt_available">true</bool>
````
Also ramdisk side do you define the ims services?
````bash
service ims_rtp_daemon /system/bin/ims_rtp_daemon
   class main
   user system
   socket ims_rtpd stream 0660 system radio
   group radio net_raw diag diag inet log
   disabled

on property:sys.ims.DATA_DAEMON_STATUS=1
    start ims_rtp_daemon

service imscmservice /system/bin/imscmservice
    class main
    user system
    group radio net_raw diag diag log
    disabled

on property:sys.ims.DATA_DAEMON_STATUS=1
    start imscmservice

service imsdatadaemon /system/bin/imsdatadaemon
    class main
    user system
    socket ims_datad stream 0660 system radio
    group system wifi radio inet net_raw log diag net_admin
    disabled

on property:sys.ims.QMI_DAEMON_STATUS=1
    start imsdatadaemon

service imsqmidaemon /system/bin/imsqmidaemon
    class main
    user system
    socket ims_qmid stream 0660 system radio
    group radio net_raw log diag
````

bhb27, 2016-10-11:
I have in the pass already unpack the boot.img to help all the services are there

"imscmservice" doesn't exist on any ROM that have VoLTE support  for this device and that files also doesn't exist in any vendor folder from Moto.

https://github.com/bhb27/boot_quark/tree/VRZ_XT1254_SU4TL-44_44/ramdisk

there is one VoLTE .rc
https://github.com/bhb27/boot_quark/blob/VRZ_XT1254_SU4TL-44_44/ramdisk/init.mmi.volte.rc
and it points to one rcs .sh
https://github.com/bhb27/boot_quark/blob/VRZ_XT1254_SU4TL-44_44/rcs_config.sh

Binaries that those files point are in the SU44/49 from Verizon you must have that already if not learn how to extract they from here there is a tutorial in XDA...
https://www.androidfilehost.com/user/?w=settings-dev-files&flid=41177
or check my vendor repo...

the flags I use to enable the "enhanced 4G LTE Mode" is this...

    
    <bool name="config_device_volte_available">true</bool>

    
    <bool name="config_carrier_volte_available">true</bool>

    
    <bool name="config_carrier_volte_provisioned">true</bool>

    
    <bool name="config_device_vt_available">true</bool>

    
    <bool name="config_carrier_vt_available">true</bool>

    
    <bool name="config_device_wfc_ims_available">true</bool>

    
    <bool name="config_carrier_wfc_ims_available">true</bool>

have fun if any helps let me know..

vish555, 2017-03-03:
Did anyone get this thing work!!???

awheeler-mmi, 2017-09-27:
This isn't something we support, but we're happy that folks are experimenting!  I'm closing this Issues since there hasn't been activity in a while.

closed 2017-09-27 by awheeler-mmi