MotorolaMobilityLLC/kernel-msm #86: Latest XT1078 release [ MPBS24.65-34-5 ]

luk1337, 2016-10-07:
Can we get kernel source released for it?

Build prop:
```
# begin build properties
# autogenerated by buildinfo.sh
ro.build.id=MPBS24.65-34-5
ro.build.display.id=MPBS24.65-34-5
ro.build.version.incremental=7
ro.build.version.sdk=23
ro.build.version.preview_sdk=0
ro.build.version.codename=REL
ro.build.version.all_codenames=REL
ro.build.version.release=6.0
ro.build.version.security_patch=2016-08-01
ro.build.version.base_os=motorola/thea_retbr_ds/thea_umtsds:6.0/MPB24.65-34/29:user/release-keys
ro.mot.base_buildid=MPB24.65-34/29
ro.build.date=Sat Sep 10 01:53:11 CDT 2016
ro.build.date.utc=1473490391
ro.build.type=user
ro.build.user=hudsoncm
ro.build.host=ilclbld29
ro.build.tags=release-keys
ro.build.flavor=thea_retbr_ds-user
ro.product.model=XT1078
ro.product.brand=motorola
ro.product.name=thea_retbr_ds
ro.product.device=thea_umtsds
ro.product.board=MSM8226
# ro.product.cpu.abi and ro.product.cpu.abi2 are obsolete,
# use ro.product.cpu.abilist instead.
ro.product.cpu.abi=armeabi-v7a
ro.product.cpu.abi2=armeabi
ro.product.cpu.abilist=armeabi-v7a,armeabi
ro.product.cpu.abilist32=armeabi-v7a,armeabi
ro.product.cpu.abilist64=
ro.product.manufacturer=motorola
ro.product.locale=pt-BR
ro.wifi.channels=
ro.board.platform=msm8226
# ro.build.product is obsolete; use ro.product.device
ro.build.product=thea_umtsds
# Do not try to parse description, fingerprint, or thumbprint
ro.build.description=thea_retbr_ds-user 6.0 MPBS24.65-34-5 7 release-keys
ro.build.fingerprint=motorola/thea_retbr_ds/thea_umtsds:6.0/MPBS24.65-34-5/7:user/release-keys
ro.build.characteristics=default
# end build properties
#
# from device/moto/thea_umtsds/system.prop
#
#
# system.prop for msm8226
#

# Use reference RIL for initial bringup
#rild.libpath=/system/lib/libreference-ril.so
rild.libpath=/vendor/lib/libril-qc-qmi-1.so
rild.libargs=-d /dev/smd0
persist.radio.nitz_plmn=
persist.radio.nitz_lons_0=
persist.radio.nitz_lons_1=
persist.radio.nitz_lons_2=
persist.radio.nitz_lons_3=
persist.radio.nitz_sons_0=
persist.radio.nitz_sons_1=
persist.radio.nitz_sons_2=
persist.radio.nitz_sons_3=
ril.subscription.types=NV,RUIM
DEVICE_PROVISIONED=1
persist.radio.msgtunnel.start=false

#
# system props for the cne module
#
persist.cne.feature=0

# Skip /sys/power/wait_for_fb_* nodes and
# force FB to be always on
debug.sf.fb_always_on=1

debug.sf.hw=1
debug.egl.hw=1
debug.composition.type=c2d
persist.hwc.mdpcomp.enable=true
debug.mdpcomp.logs=0
dalvik.vm.heapsize=36m
dev.pm.dyn_samplingrate=1

persist.demo.hdmirotationlock=false
ro.hdmi.enable=true
tunnel.decode=false
tunnel.audiovideo.decode=false
lpa.decode=true
lpa.use-stagefright=true

#system props for the MM modules

media.stagefright.enable-player=true
media.stagefright.enable-http=true
media.stagefright.enable-aac=true
media.stagefright.enable-qcp=true
media.stagefright.enable-fma2dp=true
media.stagefright.enable-scan=true
mmp.enable.3g2=true
#9273 is decimal sum of supported codecs in AAL
#codecs:(PARSER_)AVI AC3 ASF AAC QCP DTS 3G2 MP2TS
mm.enable.qcom_parser=37491

# VIDC: debug_levels
# 1:ERROR 2:HIGH 4:LOW 0:NOlogs 7:AllLogs
vidc.debug.level=1

#
# system props for the data modules
#
ro.use_data_netmgrd=true

#system props for time-services
persist.timed.enable=true

#
# system prop for opengles version
#
# 196608 is decimal for 0x30000 to report version 3
ro.opengles.version=196608

#
# System props for telephony
# System prop to turn on CdmaLTEPhone always
telephony.lteOnCdmaDevice=0

#
# enable esdfs by default
#
persist.esdfs_sdcard=true

#
#snapdragon value add features
#
# System props for audio
ro.qc.sdk.audio.ssr=false
ro.config.vc_call_vol_steps=8
# FM maximum volume be between 0 and 8192 (0dB)
ro.audio.fm_max_volume=4096

#need rotation info in audio-land
ro.audio.monitorRotation=1

# Enable/disable cabl
ro.qualcomm.cabl=0

# system prop for NFC DT
ro.nfc.port=I2C

#
# system property for Bluetooth hci transport
#
ro.qualcomm.bt.hci_transport=smd

#
# system property for Bluetooth HFP version
#
ro.bluetooth.hfp.ver=1.6

#
# system property for Bluetooth SAP support
# Motorola disabled SAP
#
ro.qualcomm.bluetooth.sap=false

# Bluetooth LE power class
qcom.bt.le_dev_pwr_class=1

##property to choose between virtual/external wfd display
persist.sys.wfd.virtual=0
# Motorola, ams087, 09/24/2013, IKJBMR2-6432 : disable tunnel audio encode
#tunnel.audio.encode = true

# Enable SSR
persist.sys.ssr.restart_level=3
persist.sys.qc.sub.rdump.on=1

# For TCMD debug level
persist.tcmd.log.limit=0

# Disable QCOM net early sockets feature
net.early.sockets=0

# Radio RIL properties
persist.radio.no_wait_for_card=1
persist.radio.apm_sim_not_pwdn=1
persist.radio.dfr_mode_set=1
persist.radio.relay_oprt_change=1

#Enable offload audio playback by default
audio.offload.disable=0

#Disable offload audio video playback by default
av.offload.enable=false

#
# ADDITIONAL_BUILD_PROPERTIES
#
ro.product.display=Moto G
ro.persist.qcapb=0
ro.build.version.qcom=AU_LINUX_ANDROID_LNX.LA.3.5.1_RB1.04.04.02.048.045
ro.build.version.full=Blur_Version.24.201.5.thea_retbr_ds.retbr.en.BR
ro.mot.build.customerid=retbr
ro.mot.build.version.sdk_int=24
ro.mot.build.product.increment=201
ro.mot.build.version.release=24.201
persist.sys.timezone=America/Sao_Paulo
persist.radio.plmn_name_cmp=1
ro.camera.sound.forced=0
ro.com.android.dataroaming=false
persist.mot.gps.assisted=false
persist.radio.process_sups_ind=1
persist.radio.all_bc_msg=all
telephony.lteOnCdmaDevice=0
ro.telephony.default_network=9
persist.radio.multisim.config=dsds
persist.radio.force_get_pref=1
dalvik.vm.boot-dex2oat-threads=2
dalvik.vm.dex2oat-threads=4
ro.MAX_HIDDEN_APPS=12
dalvik.vm.heapstartsize=8m
dalvik.vm.heapgrowthlimit=96m
dalvik.vm.heapsize=256m
dalvik.vm.heaptargetutilization=0.75
dalvik.vm.heapminfree=2m
dalvik.vm.heapmaxfree=8m
persist.sys.logkit.ctrlcode=0
ro.vendor.extension_library=/vendor/lib/libqc-opt.so
persist.radio.apm_sim_not_pwdn=1
ro.usb.mtp=0x2e82
ro.usb.mtp_adb=0x2e76
ro.usb.ptp=0x2e83
ro.usb.ptp_adb=0x2e84
ro.usb.bpt=0x2e28
ro.usb.bpt_adb=0x2e29
ro.usb.bpteth=0x2e2a
ro.usb.bpteth_adb=0x2e2b
persist.audio.calfile0=/etc/Bluetooth_cal.acdb
persist.audio.calfile1=/etc/General_cal.acdb
persist.audio.calfile2=/etc/Global_cal.acdb
persist.audio.calfile3=/etc/Handset_cal.acdb
persist.audio.calfile4=/etc/Hdmi_cal.acdb
persist.audio.calfile5=/etc/Headset_cal.acdb
persist.audio.calfile6=/etc/Speaker_cal.acdb
persist.audio.dualmic.config=endfire
persist.audio.fluence.voicecall=true
persist.audio.fluence.voicecomm=false
persist.audio.fluence.voicerec=false
persist.audio.fluence.speaker=false
ro.bug2go.magickeys=24,26
persist.sys.qc.sub.rdump.max=0
ro.adb.secure=1
ro.sf.lcd_density=320
ro.frp.pst=/dev/block/platform/msm_sdcc.1/by-name/frp
ro.config.ringtone=Moto.ogg
ro.config.notification_sound=Moto.ogg
ro.config.alarm_alert=Oxygen.ogg
persist.radio.apn_delay=5000
persist.sys.media.use-awesome=false
audio.offload.disable=0
audio.offload.pcm.16bit.enable=false
audio.offload.pcm.24bit.enable=false
audio.offload.gapless.enabled=false
av.offload.enable=false
mm.enable.qcom_parser=4643
persist.radio.mot_ecc_enabled=1
persist.radio.mot_ecc_custid=common
keyguard.no_require_sim=true
drm.service.enabled=true
mdc_initial_max_retry=10
ro.com.google.clientidbase=android-motorola
ro.com.google.clientidbase.ms=android-motorola
ro.com.google.clientidbase.am=android-motorola
ro.com.google.clientidbase.gmm=android-motorola
ro.com.google.clientidbase.yt=android-motorola
ro.url.legal=http://www.google.com/intl/%s/mobile/android/basic/phone-legal.html
ro.url.legal.android_privacy=http://www.google.com/intl/%s/mobile/android/basic/privacy.html
ro.setupwizard.mode=OPTIONAL
ro.com.google.gmsversion=6.0_r1
ro.carrier=unknown
ro.media.enc.aud.fileformat=amr
ro.media.enc.aud.codec=amrnb
persist.sys.dalvik.vm.lib.2=libart
dalvik.vm.isa.arm.variant=krait
dalvik.vm.isa.arm.features=default
net.bt.name=Android
dalvik.vm.stack-trace-file=/data/anr/traces.txt
ro.qc.sdk.izat.premium_enabled=0
ro.qc.sdk.izat.service_mask=0x0
persist.gps.qc_nlp_in_use=0
ro.gps.agps_provider=1
persist.radio.dont_use_dsd=true
persist.radio.set_dds_ia_apn=1
ro.expect.recovery_id=0x15982aff2e2ccbcccc94ffc78bbf0810feda1bef000000000000000000000000

```

myzb, 2016-11-27:
Any news on this?

charleseb, 2016-11-28:
Hi Matthias,

We’re working to release this version. It will be released in a few days.

Thanks,
Charles

charleseb, 2017-01-20:
Hi @myzb,

Please check at:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-MPB24.65-34

Thanks,
Charles

closed 2017-01-20 by charleseb