MotorolaMobilityLLC/kernel-msm #87: when release kernel source for Moto z play?

wzsx150, 2016-10-21:
hi,
would you release the kernel source for Moto z play?
the version is motorola/addison_retcn/addison:6.0.1/MCN24.104-35.2/1:user/release-keys

thank you very much!

charleseb, 2016-10-24:
Hi @wzsx150,

I've just released https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-MPN24.104-44.
We are working to release MCN24.104-35.2 
I'll push such version in a few days.

Thanks,
Charles

charleseb, 2016-10-25:
Hi @wzsx150,

This version is in our pipeline to be released.
I'll release this version in a few days.

----
Thanks,
Charles

wzsx150, 2016-10-25:
@charleseb thanks.

charleseb, 2016-11-07:
Hi @wzsx150,

The version MCN24.104-35.2 has been released.
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-MCN24.104-35.2

Thanks,
Charles

closed 2016-11-07 by charleseb