MotorolaMobilityLLC/kernel-msm #90: VOLTE

Subinsmani, 2016-10-29:
Please give source code to enable volte feature for moto g 2015 osprey.

luca020400, 2016-10-29:
Short explanation:
You can't magically enable VOLTE
sim salabim doesn't work :(
Long explanation:
VOLTE needs software and hardware support ( modem/band )
Moto can't simply release the code of the modem and therefore unless your device has already HW support you can't enable VOLTE
You can see another issue about VOLTE here #85 and try to enable it by yourself

Subinsmani, 2016-10-29:
But as of now we need volte Support for India,,, Its running only in Stock android but on custom rom it dosnt run,,
Because we didnt get android 7 officially from motorola so we also need android n feature with VoLTE Feature.

luca020400, 2016-10-29:
Flash the stock ROM or wait for someone to fix it in the custom ROMs
Motorola has nothing to do with that

Subinsmani, 2016-11-14:
But motorola should give solution for volte support...
U all are not giving nougat so,,, so we installed custome rome which is much smooter then stock rom... But only thing volte is not ennable. U should give the source to enable only volte feature... It is heighly demand for indian customer, so give us solution volte support... Xt1550...
Please give ussolution for volte

Subinsmani, 2016-11-15:
?????

Subinsmani, 2016-11-17:
hello???? i need the solution,,,,

Subinsmani, 2016-11-17:
m not spamming here,,, I want solution to enable volte feature on custom rom,,,,
Give guide to enable,,

DevriesL, 2016-11-17:
VOLTE IS RELATED TO HARDWARE AND BASEBAND SOFTWARE.
NO GUIDE OR OPENSOURCE SOFTWARE CAN DO THIS.

luca020400, 2016-11-17:
Here we go again ...

closed 2016-11-17 by Subinsmani