MotorolaMobilityLLC/kernel-msm #92: Share the source code of VoLTE Of Moto G 2015 (Osprey)

kulankit97, 2016-11-18:
Hi, 
Please share the source code of VoLTE Of Moto G 2015 (osprey) to implement enhanced 4G ( calls over LTE network).. Enhanced 4G (calls over LTE) is needed because of the new service provider JIO.. It works perfectly fine on Stock Android, but doens't work on Custom ROMs like Cyanogenmod.. I asked the maintainer to get implemented VoLTE in Cyanogenmod rom.. but the source code is not shared by Motorola with us to implement VoLTE in custom roms.. please share and help us..

luca020400, 2016-11-18:
Come on ... Not again ...

kulankit97, 2016-11-18:
This is the most wanted feature for us..

kulankit97, 2016-11-18:
Ok.. tell me the technical difficulties.. we will sort out..

luca020400, 2016-11-18:
Or maybe just check the other issues about VOLTE? #90 for example

kulankit97, 2016-11-18:
Who can implement VoLTE feature in custom rom? Rom mainter or Motorola?

luca020400, 2016-11-18:
Does stock support volte > yes it can be implemented in custom roms ( maybe, it doesn't always work )
Stock doesn't support volte > you can't

kulankit97, 2016-11-18:
Stock supports volte > yes it can be implemented in custom roms ( maybe, it doesn't always work )

Then why devs are not working on it?

luca020400, 2016-11-18:
Maybe they tried and it didn't work? Motorola may be using custom libs/apks that aren't compatible with the open source Android framework
Or maybe it doesn't just work?

If you really want volte try to add it by yourself

kulankit97, 2016-11-18:
I don't know how to add it.. I need someone who knows it..

luca020400, 2016-11-18:
What if they tried and failed? We "devs" aren't magically able to fix everything

kulankit97, 2016-11-18:
They should troubleshoot the issue as it is the most important topic in India..

luca020400, 2016-11-18:
You know that we don't work for you?
We work on custom roms in our free time as a hobby

Anyway close this issue

somcom3x, 2016-11-18:
I guess GitHub is no longer our safe haven...

kulankit97, 2016-11-18:
I know and I respect all devs who work for us.. But the issue is also important.. please understand it.. anyways I am closing this issue now..

closed 2016-11-18 by kulankit97

luca020400, 2016-11-18:
@somcom3x sadly, it isn't anymore :(