MotorolaMobilityLLC/kernel-msm #96: Kernel Relseased request for latest Quark Moto Maxx (MPGS24.107-70.2-2)

bhb27, 2016-12-06:
Please released the source for latest Marshmallow ROM for the Moto Maxx

	ro.build.id=MPGS24.107-70.2-2
	ro.build.display.id=MPGS24.107-70.2-2
	ro.build.version.security_patch=2016-09-01
	ro.build.version.base_os=motorola/quark_retbr/quark_umts:6.0.1/MPG24.107-70.2/2:user/release-keys
	ro.mot.base_buildid=MPG24.107-70.2/2
	ro.build.description=quark_retbr-user 6.0.1 MPGS24.107-70.2-2 2 release-keys
	ro.build.fingerprint=motorola/quark_retbr/quark_umts:6.0.1/MPGS24.107-70.2-2/2:user/release-keys

thanks.

charleseb, 2017-01-20:
Hi @bhb27,

This version is already released.
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-MPG24.107-70.2

Thanks,
Charles

bhb27, 2017-01-21:
Hi @charleseb 
That can be correct, I'm asking for **MPGS24.107-70.2-2**.
I'm asking for the kernel source for the OS released in October/November of 2016, that kernel source you are pointing is from the OS released back in May.

In the OTA pack of MPGS24.107-70.2-2 there is updates to the boot.img and to the wifi driver (qca_cld_wlan.ko qcacld-2.0), and that OTA was released to be install over MPG24.107-70.2, so must have changes there.

I believed that source has updates to fs (f2fs released separated by Motorola), the wifi driver (qcacld-2.0 released separated by Motorola), security updates and bug fix up to that date.

So please check and do released that for me.

charleseb, 2017-01-25:
Oh ok @bhb27, I'm working to release this version.
Sorry about that.

Charles

bhb27, 2017-01-25:
Not a problem @charleseb thanks.

charleseb, 2017-01-27:
Version MMI-MPGS24.107-70.2 released.

Thanks,
Charles

closed 2017-01-27 by charleseb

bhb27, 2017-01-27:
@charleseb thanks for this and #97.

:+1: