MotorolaMobilityLLC/kernel-msm #98: Moto G4/G4 Plus - NPJ25.93-11 Kernel Sources

Vachounet, 2016-12-08:
Now it's started rolling out, could you provide the corresponding sources ?

@charleseb

luk1337, 2016-12-20:
-1

luca020400, 2016-12-20:
-2

charleseb, 2016-12-20:
Hi all,
I've started the process to release this version. It will be available in a few days.

Thanks,
Charles

xndres, 2017-01-16:
One month and source is not ready ...

rahulsnair, 2017-01-16:
@charleseb  it been one month after your "few days"  It feels really sad that we need to keep reminding about the source request where you should have already published the source without request. I also wonder how you managed to release the kernel source for griffin within a week after the issue has been opened. Is motorola releasing kernel source based on price tag?

somcom3x, 2017-01-16:
Most of you seem to be forgetting the fact that most of Motorola's development team is gone now...
I'm fairly surprised the development team at Motorola still bothers replying to these toxic mobs. 

Why not read up on issues #1, #4, #62 and #86? You simply ask once, and wait.

And to touch off on @rahulsnair 's price-point comment, wouldn't the force be the first to get the kernel released with that logic? A lot of you are being very unprofessional with comments like this.

If you want it quick, what's stopping them from releasing the kernel with 0 git history? I strongly prefer moderately slow releases with actual history instead of guessing what revision and changes the kernel has had.

“Patience is bitter, but its fruit is sweet.” 
― Aristotle

The release is most likely right around the corner...

charleseb, 2017-01-18:
Hi all,
MMI-NPJ25.93-11 released.
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NPJ25.93-11

closed 2017-01-18 by charleseb