MotorolaMobilityLLC/kernel-msm #99: Moto Z(griffin)- NCC25.106-11 kernel sources

schqiushui, 2017-01-05:
Could u release kernel sources? 
@charleseb

somcom3x, 2017-01-12:
Its pretty pointless to keep spamming +1's. I'm sure it will be released soon.

charleseb, 2017-01-12:
Hi all,

I'm working to release Moto Z code, it will be available in a few days.

Thanks,
Charles

Obter o Outlook para Android<https://aka.ms/ghei36>

bhb27, 2017-01-12:
@charleseb **Way the preference to some devices**?

I open more then one kernel request and I never get a answer, way is that?

The big Way is... **Way we even need to request? Way you don't released the source after the SW has be pushed to the devices like others vendors do?**

Open over 30 days ago no answer those are just one line below this request in the list
https://github.com/MotorolaMobilityLLC/kernel-msm/issues/96
https://github.com/MotorolaMobilityLLC/kernel-msm/issues/97

Can you guys check the list of open request here.
https://github.com/MotorolaMobilityLLC/kernel-msm/issues

That kernel source I ask is just a update takes no time to do it, plus the SW for those kernel was released way more time ago to the devices.

**Motorola should release the source for they open source kernel and others projects without any request, feels really disrespectful, come here and never have an answer, and those are not the first I open and did not received a answer plus many other that are in the same situation.**

Please don't forget that we here asking are not random github user we are yours customers, be more polite to yours customers.
Thank you.

luk1337, 2017-01-12:
@bhb27 just a month? You're being impatient lmao, see #86.

charleseb, 2017-01-12:
Hi @bhb27 an @luk1337,

We have a backlog of versions to release and we have some processes to conclude before to release a version, that's why we release some products before, but without preference.
Sometimes is not possible to reply all requests here in kernel project issues session and others in a short time.
Sorry if we not replied you requests yet, we are working to release the code for all versions available.
I can see your requests on the queue and I'm working to release it as soon as possible.

Thanks,
Charles

charleseb, 2017-01-13:
Hi all,

For the time being, I've just released version MMI-NPL25.86-15 related to Griffin (Nougat).
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NPL25.86-15
I'm working to release other versions as soon as possible.

Thanks,
Charles

schqiushui, 2017-01-13:
Thank U very very very much!!!

Vachounet, 2017-01-26:
Could you provide an "How to build" for this ?

ghazel, 2017-03-26:
Which defconfig should be used?

charleseb, 2017-03-27:
Hi @Vachounet,

Please use this dtbtool:
https://github.com/MotorolaMobilityLLC/readme/tree/master/dtbtool_MMI-NPL25.86-15

Thanks,
Charles

Vachounet, 2017-03-31:
@ghazel msm_defconfig + moto-msm.config + moto-camera-griffin.config
From arch/arm64/configs/ && arch/arm64/configs/ext_config/

charleseb, 2017-03-31:
Hi @Vachounet and @ghazel,

Please check this readme file:
https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-NPL25.86-15.txt

Thanks,
Charles

chenyt9, 2017-08-09:
Hi @all,
@schqiushui 

Version NCC25.106-11 released.
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NCC25.106-11

Thanks,

luca020400, 2017-08-09:
Just 8 months for a release.

schqiushui, 2017-08-27:
thanks a lot @chenyt9 for responses.

closed 2017-08-27 by schqiushui