MotorolaMobilityLLC/kernel-msm #104: GPL Source Request for Moto G 5 Plus (potter)

RobGries, 2017-04-03:
Hello,

I have developer unlocked my Moto G 5 Plus(potter), and I'd like to obtain the GPL sources for this model so that I can build my own kernel.

Regards,

Rob

rayzen6, 2017-04-04:
+1 plz release it as soon as possible.

chenyt9, 2017-04-05:
Sorry, the job is in the process, we need a few days to prepare, please wait. Thank you!

rayzen6, 2017-04-05:
Hi @chenyt9

Thank you for your quick response. Take your time. We'll wait. :)

Vachounet, 2017-04-08:
FYI, addison sources can be used to build potter kernel.

RobGries, 2017-04-08:
Interesting, so do I just build for addison and use it on potter, or are there a few modifications to be made? Are the device trees the same?

luca020400, 2017-04-08:
@RobGries the kernel is shared, all you have to do is build with the potter defconfig

RobGries, 2017-04-08:
Gotcha, so the same build notes for addison apply here too, just sub in a potter specific defconfig and it should be good to go?
https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-NPN25.137-15.txt

luca020400, 2017-04-08:
@RobGries yes basically all you have to do is adjust this command for potter
````
perl -le 'print \"# This file was automatically generated from:\\n#\\t\" . join(\"\\n#\\t\", @ARGV) . \"\\n\"' kernel/arch/arm/configs/msmcortex-perf_defconfig kernel/arch/arm/configs/ext_config/moto-msmcortex.config kernel/arch/arm/configs/ext_config/debug-msmcortex.config kernel/arch/arm/configs/ext_config/moto-camera-addison.config kernel/arch/arm/configs/ext_config/moto-addison.config && cat kernel/arch/arm/configs/msmcortex-perf_defconfig kernel/arch/arm/configs/ext_config/moto-msmcortex.config kernel/arch/arm/configs/ext_config/debug-msmcortex.config kernel/arch/arm/configs/ext_config/moto-camera-addison.config kernel/arch/arm/configs/ext_config/moto-addison.config ) > $kernel_out_dir/mapphone_defconfig || ( rm -f $kernel_out_dir/mapphone_defconfig && false )
````

HostZero, 2017-04-09:
@luca020400

I've got my G5 Plus hands on to. I am waiting for GPL kernel sources. I have seen here addison kernel sources can be used yes, I extracted the Potter_defconfig. So should I just build it? Or should I run the Perl command on kernel directory and then build?

luca020400, 2017-04-09:
If you have extracted the config you could just use it

HostZero, 2017-04-09:
Thx. I will try my PC is blocked right now once I get on I will update with the information
Thank you

Vachounet, 2017-04-09:
@HostZero Yeah that's what i did, extracted defconfig from stock boot using extract-ikconfig, then cleanup a bit debug flags.

HostZero, 2017-04-09:
I have sent my workstation to office for upgrade. Once I reinstall Linux and all I will thank you a lot actually for helping me @Vachounet

HostZero, 2017-04-11:
@Vachounet I have tried to compile.. with stock lineage 4.9 gcc toolchain (arm) where as i got some this kind of log.

I had to downgraded my Moto to 35 to 33 update as my device was bricked and i could only find 33 firmware.

defconfig i used: https://pastebin.com/8nDi5ft6

error i got while compiling kernel: https://pastebin.com/w3DRs4SF

HostZero, 2017-04-11:
Update: I started building with a [WIP] Device Tree
Device: https://github.com/HostZero/android_device_motorola_potter/tree/cm-14.1
https://github.com/HosrZero/android_kernel_motorola_msm8953/tree/cm-14.1

With the defconfig mentioned above.

The build is stuck at 99% Target Ramdisk.

Vachounet, 2017-04-11:
@HostZero You'll have those errors even with official release (all latest device sources has, you have to fix them by yourself).
Your defconfig seems good. Also, having a different file size doesn't mean it will not work.

HostZero, 2017-04-11:
@Vachounet those kernel issues seemed to be fixed while building I'm trying to build lineage and test. I tried to boot but it couldn't it was stuck on loop.

Vachounet, 2017-04-11:
@HostZero That's a good start :)

HostZero, 2017-04-11:
@Vachounet I have a last question. I took the addition proprietary list and extracted I couldn't find many blobs even stock hw blobs? Is it fine or I must search for another alternative blob?

Vachounet, 2017-04-11:
The whole vendor tree has to be updated for potter.

HostZero, 2017-04-11:
Any suggestions like how to find the required blobs quickly?

Vachounet, 2017-04-11:
You can use this as based. Just pushed my last changes.
https://github.com/boulzordev/android_vendor_motorola_potter

HostZero, 2017-04-11:
Thx a lot I will try with that and inform after the build is successful. I will not forget your help.

Vachounet, 2017-04-11:
Already got a complete build, but got an error during flash. :/ Let me know how it goes for you.

HostZero, 2017-04-11:
Sure. I will try and inform you here

chenyt9, 2017-04-11:
Hi all,

I've just released version MMI-NPN25.137-33 related to Potter (Nougat).
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NPN25.137-33

Thanks.
Yuting

chenyt9, 2017-04-11:
@rayzen6 @RobGries  Please fetch the code  from：
 https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NPN25.137-33

HostZero, 2017-04-11:
@chenyt9 thanks for informing us. Can you please name the defconfig we must use?

chenyt9, 2017-04-11:
@all,  Here is readme file : https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-NPN25.137-33
@HostZero, You can cat kernel/arch/arm/configs/msmcortex-perf_defconfig kernel/arch/arm/configs/ext_config/moto-msmcortex.config kernel/arch/arm/configs/ext_config/mot8953-potter.config >.config

HostZero, 2017-04-11:
@Vachounet I have a question i am facing a issue while compiling 

Communication error with Jack server (56). Try 'jack-diagnose'

how do i fix this?

luca020400, 2017-04-11:
@HostZero probably the jack server is running out of mem, increase max mem size
https://github.com/LineageOS/android_vendor_cm/commit/84fda60f6be9a1c9e072ef6a335fa6300bd6a570
export JACK_SERVER_VM_ARGUMENTS="-Xmx=4GB" for example

HostZero, 2017-04-11:
@luca020400 i have exported some arguments later i got a new error like their is no clue like how to sort out

I exported this: export JACK_SERVER_VM_ARGUMENTS="-Dfile.encoding=UTF-4 -XX:+TieredCompilation -Xmx3000m"

I got a error like this: https://pastebin.com/JuqeZX3J

Luca, can i contact you in a different way as Moto would feel this uncomfortable.

Jack errors are the errors i find all the time. shall i remove that export and export the argument provided by you?

Regards,
Haneesh Raja

luca020400, 2017-04-11:
As long we aren't /really/ offtopic it's fine
All I can say it's that 4GB of ram may not be enough and jack is a bitch

HostZero, 2017-04-11:
@luca020400 is their any other way so I can run the build? I'm on Ubuntu 16.04. any other way? I tried to get a 6gb dam where as my mother board is a old one. Till I find a better mother board can I not run the build?

HostZero, 2017-04-12:
@luca020400 

[ERROR]:
6 warnings generated.
[ 38% 13406/34942] Ensure Jack server is installed and started
Jack server already installed in "/home/mickey/.jack-server"
Server is already running
[ 38% 13415/34942] Building with Jack: /home/mickey/lineage-14.1/.../obj/JAVA_LIBRARIES/core-all_intermediates/with-local/classes.dex
FAILED: /bin/bash /home/mickey/lineage-14.1/out/target/common/obj/JAVA_LIBRARIES/core-all_intermediates/with-local/classes.dex.rsp
mv: cannot stat '/home/mickey/lineage-14.1/out/target/common/obj/JAVA_LIBRARIES/core-all_intermediates/jack-rsc/classes*.dex': No such file or directory
[ 38% 13415/34942] host Java: bouncycastle-host (/home/mickey/lin...ommon/obj/JAVA_LIBRARIES/bouncycastle-host_intermediates/classes)
warning: [options] bootstrap class path not set in conjunction with -source 1.7
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
1 warning
ninja: build stopped: subcommand failed.
build/core/ninja.mk:151: recipe for target 'ninja_wrapper' failed
make: *** [ninja_wrapper] Error 1
make: Leaving directory '/home/mickey/lineage-14.1'

#### make failed to build some targets (40:34 (mm:ss)) ####
[/ERROR]

I can find a file is missing. any repository is missing?

faiyazsheth, 2017-04-12:
@HostZero

Check your XDA PM mate. :)

jaggu1989, 2017-04-15:
@hostzero .. I too got the same error ... With Jack .... I also have 4 gb ram ... So the culprit is ram

jaggu1989, 2017-04-15:
@HostZero
 .. I too got the same error ... With Jack .... I also have 4 gb ram ... So the culprit is ram

charleseb, 2017-06-05:
Hi all,

Version MMI-NPN25.137-67 released.
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NPN25.137-67

closed 2017-06-05 by charleseb