MotorolaMobilityLLC/kernel-msm #106: Phantom touches on Moto G4(2016)

zoggn, 2017-04-04:
When mobile phone charged, touchscreen becoming very Crazy(phantom touches on all screen). Using stock charger!

DaviAntonio, 2017-09-03:
I had to return one Moto G4 because its ghost touch rendered the phone unusable, even when it was not charging. I have got another Moto G4 which has the same problems, but only when charging, and the problem happens in any charging method (official charger, other chargers or USB on a laptop).

awheeler-mmi, 2017-09-27:
Hey @zoggn and @DaviAntonio, I'm sorry to hear you guys have experienced ghost touches.  Product support is [here](https://mobilesupport.lenovo.com/us/en/).  They also have a support forum that is actively monitored.  FWIW, I would try an FDR -- if you can deal with wiping your data.

closed 2017-09-27 by awheeler-mmi