MotorolaMobilityLLC/kernel-msm #109: Moto G5 Plus - NPN25.137-35 Kernel Sources

attack11, 2017-04-26:
Could you provide the corresponding sources ?

@charleseb   @chenyt9

Thanks in advance!

attack11, 2017-05-01:
Any news on this? @charleseb @chenyt9 ?

flar2, 2017-05-29:
The camera doesn't work with the potter sources posted here.  The camera works with the addison source.  Not sure if there are any other problems with the potter source code, but it would be nice if the most recent and working source were posted.

attack11, 2017-05-29:
@flar2 yes i use latest cedric sources todo build for potter stock ROM, just 2 commits more than Addison sources, sadly as you can see nobody respond here, PS: Im very happy that a dev with your talent  Will support this device :)

charleseb, 2017-05-29:
Hi @attack11,
Sorry, I've lost this thread in my inbox. That was our last released tag
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NPN25.137-33
I can see we've a more updated version to release, I'm working to release this version soon.

Thanks,
Charles

attack11, 2017-05-29:
Thanks ! @charleseb , Waiting for it !

Vachounet, 2017-05-31:
Even with Addison sources, camera doesn't work anymore with last update. (NPN25.137-67/75)

attack11, 2017-05-31:
Yes @charleseb a new version is out as pointed by @Vachounet , The build NPN25.137-67/75 , That with current kernel sources breaks camera,Please post the latest sources as soon as you can, or we are stuck with a not working camera

charleseb, 2017-06-05:
Hi all,

Version NPN25.137-67 released.
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NPN25.137-67

Thanks,
Charles

closed 2017-06-05 by charleseb

cheetos93, 2017-08-22:
Hi guy,
Im sorry I'm new with this stuff can you tell me how to use this kernel? I tried to install it with adb in twrp but an error shows saying there's a problem with the files in the zip

jobinrjohnson, 2017-08-23:
@cheetos93 this is the source code you have to build to make it flashable. It's better to ask these questions on xda or some other forum.

_Sent from my Motorola Moto G (5) using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_