MotorolaMobilityLLC/kernel-msm #112: defconfig for moto g4 plus [Bootloop]

pradeeptewary, 2017-05-12:
I tried compiling the kernel _noughat-7.0.0-release-athene_ with toolchain _arm-eabi 4.8_ and successfully generated the _zImage_ using defconfig file _msm8226_defconfig_ appended with _debug-msm8226.config_. However after merging the zImage in device boot image, the device went into _bootloop_. How to resolve the issue?

luca020400, 2017-05-12:
No wonder it doesn't boot with msm8226_defconfig lol
Use msm8952_defconfig

FYI
msm8226 = snapdragon 400
msm8952 = snapdragon 617

pradeeptewary, 2017-05-12:
thanks @luca020400 . Sorry, but i couldn't find msm8952 defconfig in https://github.com/MotorolaMobilityLLC/kernel-msm/tree/nougat-7.0.0-release-athene/arch/arm/configs . Please tell which config to use?

luca020400, 2017-05-12:
You likely need to combine msm8916-perf_defconfig and ext_config/
moto-msm8916.config and maybe other configs
@charleseb mind uploading the readme?

pradeeptewary, 2017-05-15:
I am still not able to come out of bootloop after following the above configs. I believe the zImage is not right, because defconfig file is not right. @charleseb Please help me pointing out right defconfig for moto g4 plus.

ramsudharsan, 2017-05-15:
@pradeeptewary You can still try to obtain the defconfig from the present running kernel on your device by using extract-ikconfig script

charleseb, 2017-05-15:
Hi @pradeeptewary,

Please inform the tag you're building.

Thanks,
Charles

nikitsharma15, 2017-06-08:
@pradeeptewary Run this command in terminal
./extract-ikconfig zImage > athene_defconfig

closed 2017-06-09 by pradeeptewary