MotorolaMobilityLLC/kernel-msm #116: Moto Pure XT 1575 bootloader re-lock issue

dj-4war, 2017-07-03:
I am trying to re-locking the my Moto Pure XT 1575,  I ran.

fastboot oem lock begin
fastboot oem lock   - then says that 
_Please flash valid signed images just after lock begin_ 

ok, then tried to load the boot.img(attached one ) first. 

Then it said 

_(bootloader) Please flash valid signed images just after lock begin 
(bootloader) command!_

I have downloaded image from 
https://firmware.center/firmware/Motorola/Moto%20G%20%281st%20gen-2013%29/Stock/XT1034/

Also I have tried from Motorola site too.. but did not work 
https://motorola-global-portal.custhelp.com/app/standalone/bootloader/recovery-images#

Please help. If I m using wrong image please point me to right boot image. 

All I want is to lock my Bootloader.

[boot.img.zip](https://github.com/MotorolaMobilityLLC/kernel-msm/files/1119702/boot.img.zip)

luca020400, 2017-07-03:
fastboot oem lock begin
flash full stock firmware
fastboot oem lock

dj-4war, 2017-07-03:
Thanks for the quick response. When I try. 
 flash full stock firmware
-bash: flash: command not found

r3pwn, 2017-07-03:
@luca020400 wasn't saying to run the command "flash full stock firmware" (because, it's not a command). He was saying to flash the full stock firmware via fastboot. Just flash each file in that firmware package you downloaded earlier (the one from firmware.center) to its corresponding partition. For example, system.img would get flashed to the system partition (fastboot flash system.img system), boot.img would get flashed to the boot partition (fastboot flash boot.img boot), and do that for the rest of the images in that package.
Additionally, this isn't the place to report issues with your device. This is for developers who have issues with the source code Motorola provides. You would probably be better suited over at the XDA-Developer forums for your specific device.
I just noticed you downloaded the wrong package for your device, too. Make sure that the XT model number for the package you download matches *EXACTLY* the same number your device has. If it's a different number, you could permanently damage your device.

dj-4war, 2017-07-04:
Thanks for the reply, 

I tried flashing the full stock, with the official image.
https://motorola-global-portal.custhelp.com/app/standalone/bootloader/recovery-images
This is not a device specific and official flagship ROM where I never flashed before. 

When I try with the above. I faced below issues. 

1. When I fired fastboot flash partition gpt.bin 
It said partetion issue, when I looked at some threads they say that this is due to gpt bin upgraded or degraded version. Picked some other GPT from image available in firmware center. It went through successfully .
2. Then I flashed individual package from official ROM but did not work.

Please help me here. I felt rather XDA as you are the official Moto Devs I thought of  contacting you.

r3pwn, 2017-07-05:
The GPT partition map doesn't usually need to be re-flashed. My apologies for not mentioning that earlier. Just flash boot, recovery, system, and run "fastboot format userdata" and "fastboot format cache".
I'm not an official Moto dev, I'm just some random guy from the internet who felt like stopping to try to help you out.

awheeler-mmi, 2017-09-20:
Hey dj-4war,

I know this thread is dated, but I am responding for others that may be in a similar situation.  Looks like you tried to flash images for a Moto G onto a Moto X. This won't actually work, and could lead to your device getting into a corrupted state.

I definitely appreciate all the Motorola device hacking that goes on!  But, sadly, we don't offer much technical support on doing it.   XDA is a great site for hacking!

Thanks,

-andrew

closed 2017-09-20 by awheeler-mmi