MotorolaMobilityLLC/kernel-msm #117: Moto Z(griffin)-NPL26.118-20 kernel sources

schqiushui, 2017-07-20:
Could u please release kernel sources?

Codename: griffin
Build.No:NPL26.118-20
Android Version: 7.1.1

@charleseb

schqiushui, 2017-08-04:
NCC26.118-19(China version) is also released, could u please released the kernel sources?

@charleseb

charleseb, 2017-08-07:
Hi .古塵.,

I’m working to release this version soon.

Thanks,
Charles

chenyt9, 2017-08-29:
Hi @ALL,
@schqiushui

Version NCC25.106-11 released.
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NCC26.118-19

Thanks,

schqiushui, 2017-08-30:
camera doesn't work with this sources. I make sure set CONFIG_MSMB_CAMERA=n && CONFIG_MSMB_CAMERA_2016=y. is there anyone who make camera working?@ALL  

besides, if build greybus, these below are not defined:
"POWER_SUPPLY_PTP_POWER_SOURCE_NONE_TRUBO”
"POWER_SUPPLY_PTP_POWER_SOURCE_BATTERY_TURBO"
"POWER_SUPPLY_PTP_POWER_SOURCE_WIRED_TURBO"
"POWER_SUPPLY_PTP_POWER_SOURCE_WIRELESS_TURBO"

@chenyt9 @charleseb

erfanoabdi, 2017-08-30:
same here, 
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NCC26.118-19
is same as https://github.com/MotorolaMobilityLLC/kernel-msm/tree/nougat-7.0.0-release-CEDRIC
:( 

https://github.com/MotorolaMobilityLLC/kernel-msm/compare/MMI-NCC26.118-19...nougat-7.0.0-release-CEDRIC

but greybus and qcacld drivers are updated

chenyt9, 2017-09-12:
@schqiushui Do you follow this document compiler kernel? https://github.com/MotorolaMobilityLLC/readme/blob/master/MMI-NCC26.118-19.txt

schqiushui, 2017-09-22:
@chenyt9 
sure.

but I found working sources.

https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NPS26.116-45

this version works on Moto Z. no camera issue or other that I could find so far...

thank u all the same.

charleseb, 2017-09-22:
Hi @schqiushui,

Version MMI-NPL26.118-20 is released:
https://github.com/MotorolaMobilityLLC/kernel-msm/releases/tag/MMI-NPL26.118-20

closed 2017-09-22 by charleseb

chenyt9, 2017-09-26:
Hi，@schqiushui ，
 I have re-pushed the code of MMI-NCC26.118-19. Some of the commits is lost, please re-pull the code, thank you!
BRs.
chenyt9